dan-v/rattlesnakeos-stack #94: All packages in prebuilts repo included when specifying subset

sjdrc, 2019-01-01:
Hi Dan,

I've forked the microg repo and have included a couple of extra packages. If I only specify a subset of these packages in .rattlesnakeos.toml, the deploy confirmation lists only my specified packages for installation, but the build fails due to a filename conflict (attempting to include GmsCore and UnifiedNlp). It would seem that the build script isn't respecting the package list specified.

dan-v, 2019-01-01:
hey @sjdrc. Not sure if this is the issue, but if any of the packages you include have dependencies on the other packages, it will still pick them up regardless of if you have included them in the list or not.

sjdrc, 2019-01-01:
I've just checked the logs and the specified packages are added to the Android.mk but as all the packages are cloned into the source tree I guess android is still attempting to compile them, and that's why there's a conflict. I guess I'll just separate the different prebuilts into different repos.

closed 2019-01-01 by sjdrc