dan-v/rattlesnakeos-stack #98: Tethering Broken with GSM carrier: Ting.

L99Z0CEJER, 2019-01-10:
The changes introduced in the build_script function "patch_tethering" seem to cause tethering to fail.

With regard to the patch_tethering function, changes made in these commits (taken together or individually) result in broken tethering; specifically clients of the hotspot are unable to pass packets outside the subnet created by the hotspot.
- commit 7c235224e27225fd8cbe97fb3518dc5b992494de
- commit a51b197ec9f60182bdd0bb18ad4fb091d695af4b

Deactivating the function entirely (as in commit f99ed2a2240da990fb9680321d2a065da27ff90d ) does allow tethering to work as expected. 

Perhaps Issue #81 should be revisited?

### VERSION
Device: sailfish
Stack Version: 9.0.21
Release Channel: sailfish-stable
AOSP Build: PQ1A.190105.004
AOSP Branch: android-9.0.0_r30
Chromium Version: 71.0.3578.99
F-Droid Version: 1.5.1
F-Droid Priv Extension Version: 0.2.8

### CARRIER
Carrier: Ting GSM  (Ting is an MVNO on the T-Mobile network.)
APN: Following https://help.ting.com/hc/en-us/articles/115008890528-Activating-your-Google-Pixel#GSM
 (or the following trouble shooting section) versus the default/discovered T-Mobile APN does not seem to affect tethering capability (nor does restricting the device to 3G, etc.)

dan-v, 2019-01-10:
Thanks for the detailed report and for trying it with and without that patch - it's very useful information! 

I'm not really sure what the right approach should be at this point. I do like this change from a privacy perspective as it just makes tethering look like normal data usage, although leaking of TTL may still provide carriers with that information anyways. With or without the patch on T-Mobile USA tethering still works and this patch did fix AT&T tethering as verified by another user. But then it appears to also have broken tethering for you on Ting. 

So -- I'm not sure. I guess I could make the tether fix optional and include it in the community repo. I'll think about it a bit.

pgera, 2019-01-10:
Perhaps, the fix may be to offer additional selectable APNs ? i.e., Don't set the dun option in APNs or dun-required, but users can switch between APNs. With the t-mobile fix, there is a counterpart to it, which is that the tethering apn that they use is actually useful if you are roaming. There is some anecdotal evidence that when you are roaming, the speed limits are set depending on the APN. The default APN throttles you, but the tethering one (pcweb.tmobile.com) doesn't. I'm not sure why tethering doesn't work for ting here though, unless they use TTL or some other form of detection. @L99Z0CEJER, are you able to set TTL to -1 on your client ? Another common issue that I've heard is ipv4 v/s ipv6 weirdness.

L99Z0CEJER, 2019-02-19:
> I'm not really sure what the right approach should be at this point. I do like this change from a privacy perspective[...]

I'd suppose a universal solution may depend on the project architecture, is RattlesnakeOS an: 

- AOSP build engine + a schema for extending/modifying parts of it generally, or
- AOSP build engine + a curated set of default privacy enhancing features + a patchset application schema.

If option one is true, then this bug is against an "extension" patchset, not the AOSP build.
If option two is true, then the AOSP build is borked (albeit for tethering on Ting at least).

I'd be inclined to have a toggle ( ie --include-tethering-fix att/tmobile/ting/etc.. ) which could activate the patch_tethering function and/or selectively apply a behaviour within it, or using the community patchset mechanism for various kinds of 'tethering fixes'. It is likely that there will be increasing shenanigans with how tethering is filtered/detected by carriers; as would seem from the chatter on the topic. (imho,etc.)

> are you able to set TTL to -1 on your client ?

TTLs from (unset) -1, 1-255, via sysctl and ping -t $ttl, for USB and WIFI tethering, resulted in no change to the problematic behaviour.

### VERSION
Device: sailfish
Stack Version: 9.0.24
Release Channel: sailfish-stable
AOSP Build: PQ2A.190205.003
 AOSP Branch: android-9.0.0_r33
Chromium Version: 72.0.3626.105
F-Droid Version: 1.5.1
F-Droid Priv Extension Version: 0.2.9
Build Reason: No build is required, but FORCE_BUILD=true

### CARRIER
This was tested with multiple Ting GSM SIMs and devices.
> " If you're still having trouble with data, MMS or tethering, try switching the APN protocol and APN roaming protocol (e.g., Ting Data changes from IPv6 to IPv4 and Ting MMS changes from IPv4 to IPv6" from their website.

All combinations of IPv4/IPv6 and APN types (that the entry would accept) resulted in no change to the problematic behaviour.

dan-v, 2019-02-20:
@L99Z0CEJER thanks for this. I'm inclined to just move the change out of standard build process and into the community patch set.

commit 22547c72 (2019-03-14): * Fix issue with F-Droid latest tag not being sorted
* Remove the tethering patch as it causes issues with other carriers (closes #98). If still needed, can be ported to a custom patch.
* Add eu-north-1 region (closes #102)
* Add additional retries to Chromium build 'gclient sync'
* Update to latest Ubuntu 18.04 images
* Update README
closed 2019-03-14 by dan-v