dan-v/rattlesnakeos-stack #100: Extend custom-manifest-projects and custom-manifest-remotes

9e3fce0a, 2019-01-18:
It would be great if the `[[custom-manifest-projects]]` and `[[custom-manifest-remotes]]` customisation features were expanded to support something like this:

```toml
[[custom-manifest-remotes]]
  name = "foo"
  fetch = "https://github.com/foo/"

[[custom-manifest-projects]]
  path = "/path/to/myapps/foo/bar"
  name = "bar"
  remote = "foo"
  revision = "master"
  clone-depth = "5"
```

Currently (i.e. without this enhancement), the toml snippet above will produce the following entry in the .repo/manifest.xml file:

```xml
  <remote name="foo" fetch="https://github.com/foo/" revision="" />
   ...
  <project path="/path/to/myapps/foo/bar" name="bar" remote="foo" />
```
Maybe it should produce this instead:

```xml
  <remote name="foo" fetch="https://github.com/foo/" />
   ...
  <project path="/path/to/myapps/foo/bar" name="bar" remote="foo" revision="master" clone-depth="5" />
```

I think the places to make this enhancement is here:
- https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/templates/build_template.go#L498
- https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/templates/build_template.go#L503

9e3fce0a, 2019-01-18:
Btw, there is a workaround. If multiple projects need to reference the same remote but each with different refs, simply specify the same remote multiple times with different names.

dan-v, 2019-01-20:
@9e3fce0a - thanks for the report. Yeah, the existing support for those items are a bit limiting and it would be nice to be able to have better overall control of the resulting manifest entries. I previously thought about just allowing you to specify the full xml line instead, rather than individual properties and constructing the xml line from that. Open to ideas and suggestions on this one.

dan-v, 2021-03-01:
Next release will migrate to custom config repos: https://github.com/dan-v/rattlesnakeos-stack/#customizations.

closed 2021-03-01 by dan-v