dan-v/rattlesnakeos-stack #105: Information about how to speed up build

mariobrussi, 2019-02-12:
What is the best way to speed up build?

From your tests which is the best between 

c5.4xlarge
c5.9xlarge
c5.18xlarge

_or_ ssd version

for shortest time / lowest cost solution? How much does it cost?

There are any workaround like not compiling chromium or other?

I'm building for pixel xl and my building time is 5 hours and 11 minutes, but now i changed my key so i have to wait another 5 hours, maybe next build i can change to a more powerful instance but only if it make sense...

Thank you for your thoughts

dan-v-elevate, 2019-02-12:
I don't have any super up to date information on this, but I ran an experiment a few months ago while comparing c4 and c5 instance types of different sizes and pasted that below. Pricing is obviously going to have changed though as that's always a moving target. While not the most cost efficient, I generally build with a c5.18xlarge which takes about 2 hours for a build with Chromium and 45 mins without Chromium. Until recently c5.18xlarge was relatively cheap in the us-east-2 region but more recently has spiked quite high in price. There very well could be better options for instance type selection in terms of cost, but I just haven't investigated further. 
```
Build (without Chromium) | Build Time (Hours) | Lowest Price | Total Build Cost
c5.2xlarge | 2.5 | $0.08 | $0.19
c4.4xlarge | 2.66 | $0.14 | $0.39
c5.4xlarge | 1.75 | $0.15 | $0.27
c5.9xlarge | 1.16 | $0.34 | $0.40

Build (with Chromium)	Build Time (Hours)	Lowest Price	Total Build Cost
c5.2xlarge	9.5	$0.08	$0.74
c4.4xlarge	7.5	$0.14	$1.09
c5.4xlarge	5.25	$0.15	$0.80
c5.9xlarge	3	$0.34	$1.03
```

To save costs, after your initial build you could lock down your Chromium version by setting `chromium-version` in your config file to whatever version was built initially and doing a 'deploy' to set that config, so that future builds will be without Chromium which will significantly reduce build times.

dan-v, 2019-02-12:
I don't have any super up to date information on this, but I ran an experiment a few months ago while comparing c4 and c5 instance types of different sizes and pasted that below. Pricing is obviously going to have changed though as that's always a moving target. While not the most cost efficient, I generally build with a c5.18xlarge which takes about 2 hours for a build with Chromium and 45 mins without Chromium. Until recently c5.18xlarge was relatively cheap in the us-east-2 region but more recently has spiked quite high in price. There very well could be better options for instance type selection in terms of cost, but I just haven't investigated further.

```
Build (without Chromium)	Build Time (Hours)	Lowest Price	Total Build Cost
c5.2xlarge	2.5	$0.08	$0.19
c4.4xlarge	2.66	$0.14	$0.39
c5.4xlarge	1.75	$0.15	$0.27
c5.9xlarge	1.16	$0.34	$0.40

Build (with Chromium)	Build Time (Hours)	Lowest Price	Total Build Cost
c5.2xlarge	9.5	$0.08	$0.74
c4.4xlarge	7.5	$0.14	$1.09
c5.4xlarge	5.25	$0.15	$0.80
c5.9xlarge	3	$0.34	$1.03
```

To save costs, after your initial build you could lock down your Chromium version by setting chromium-version in your config file to whatever version was built initially and doing a 'deploy' to set that config, so that future builds will be without Chromium which will significantly reduce build times.

mariobrussi, 2019-02-12:
Thank you for your precise answer!
i've launched a c5.9xlarge and now i'll wait 3 hours teorically :)
Interesting to note that slower build cost less! in the long run maybe i can set it slow!
Maybe can be added to the README?

closed 2019-02-19 by dan-v