dan-v/rattlesnakeos-stack #107: Help to add opengapps to the build

mehdicopter, 2019-03-02:
Hello yall,
Can someone explain me how to add this repo [https://github.com/opengapps/aosp_build](https://github.com/opengapps/aosp_build) into the build ?
I read [https://github.com/dan-v/rattlesnakeos-stack#customizations](https://github.com/dan-v/rattlesnakeos-stack#customizations) but I do not know how to modify the .rattlesnakeos.toml file.
I tried this : [https://pastebin.com/Xea1VCFz](https://pastebin.com/Xea1VCFz).
I do not know how to add the variable : GAPPS_VARIANT := stock into the .mk file.

Please help :)

dan-v, 2019-03-03:
@mehdicopter - the current customizations are fairly limiting in terms of being able to easily make these required changes. This is something I want to improve in the future. 

Some notes:
1) No idea if opengapps even works with this process - I've never done it.
2) The current setup for android-prepare-vendor is only extracting the naked config and not the full config (https://github.com/anestisb/android-prepare-vendor#naked-vs-full) which I believe is needed for gapps.
3) For the other required variables and edits that you can't figure out, you could use a custom shell script (e.g. https://github.com/RattlesnakeOS/example_patch_shellscript/blob/master/00002-custom-boot-animation.sh) as a workaround.

mehdicopter, 2019-03-03:
Thanks.
I will try to build AOSP and the rest on my own then.

closed 2019-03-03 by dan-v