dan-v/rattlesnakeos-stack #109: Custom prebuilt not installed with new OTA update

60Eight, 2019-03-10:
Hi @dan-v,

I updated my config file to include a [Backup] (https://github.com/stevesoltys/backup) app. I could see in the S3 logs that this prebuilt apk is being picked up and the relevant XMLs being placed.
But after installing the OTA, I can't find the app anywhere. Do I need to flash fastboot image for this?

Any help would be great!

60Eight, 2019-03-10:
I'm using my own repo here for custom prebuilts. I just added `Backup` at the end in modules and ran the command `rattlesnakeos deploy`.

Relevant config section:
```
[[custom-prebuilts]]
  repo = "https://github.com/60Eight/ros-custom-prebuilts"
  modules = [
      "com.google.android.maps.jar",
      "FakeStore",
      "GmsCore",
      "GsfProxy",
      "MozillaNlpBackend",
      "NominatimNlpBackend",
      "Recorder",
      "Backup"
  ]

```

dan-v, 2019-03-10:
Hi @60Eight. I'm not 100% positive, but my best guess would be that you are including an unsigned APK of the Backup app and then in the the Android.mk file you are saying that it is presigned (https://github.com/60Eight/ros-custom-prebuilts/blob/master/Backup/Android.mk#L26). I think you can remove that line and it will default to the test key which should then get signed by the proper release key during the  sign_target_files_apks portion of build (https://android.googlesource.com/platform/build.git/+/master/tools/releasetools/sign_target_files_apks.py#45).

I'm also using @stevesoltys backup app in my builds, but I'm building it from source instead of a prebuilt APK. My setup looks like this:

```
[[custom-manifest-remotes]]
  name = "rattlesnakeos"
  fetch = "https://github.com/RattlesnakeOS/"
  revision = "master"

[[custom-manifest-projects]]
  path = "packages/apps/Backup"
  name = "backup"
  remote = "rattlesnakeos"
  modules = [ "Backup" ]
```

60Eight, 2019-03-10:
Thanks for the the explanation, @dan-v. I'd also like to build from source, will adding the mentioned custom manifest remotes & projects be sufficient?

dan-v, 2019-03-10:
Those two entries should be sufficient for building from source. I'd make sure to delete the Backup app folder from your repo as well or the Android.mk file will still be picked up during build and may cause issues.

60Eight, 2019-03-10:
I removed the `Backup` entry from the `modules` under `[[custom-prebuilts]]`. Should I still delete the folder from my repo?

dan-v, 2019-03-10:
Yes, still delete that Backup folder from your repo as the build process does a recursive discovery of all make files.

60Eight, 2019-03-10:
Alright, thanks for the help. I'll try it out. :+1:

60Eight, 2019-03-10:
Building from source worked. Thanks a lot, @dan-v :smiley:

closed 2019-03-10 by 60Eight