dan-v/rattlesnakeos-stack #111: Chromium build failure: gyp/lint.py tries to symlink file to itself

Rudd-O, 2019-03-13:
9.0.24.  Same env.  I added a print right before the problematic line, to print the source and the destination parameters of the `os.symlink` call -- I'll quote the print output below:

```
/home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/depot_tools/ninja -C /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium-out/ monochrome_public_apk
...
...
[1070/47073] ACTION //components/version_info/android:version_constants_java__lint(//build/toolchain/android:android_clang_arm64)
FAILED: gen/components/version_info/android/version_constants_java__lint/result.xml gen/components/version_info/android/version_constants_java__lint/config.xml 
python ../chromium/src/build/android/gyp/lint.py --lint-path ../chromium/src/third_party/android_tools/sdk/tools-lint/bin/lint --cache-dir android_lint_cache --platform-xml-path ../chromium/src/third_party/android_tools/sdk/platform-tools/api/api-versions.xml --android-sdk-version=26 --depfile gen/components/version_info/android/version_constants_java__lint.d --config-path ../chromium/src/build/android/lint/suppressions.xml --product-dir=. --processed-config-path gen/components/version_info/android/version_constants_java__lint/config.xml --result-path gen/components/version_info/android/version_constants_java__lint/result.xml --include-unexpected-failures --java-sources-file=gen/components/version_info/android/version_constants_java.sources --srcjars=@FileArg\(gen/components/version_info/android/version_constants_java.build_config:gradle:bundled_srcjars\) --can-fail-build --resource-sources=@FileArg\(gen/components/version_info/android/version_constants_java.build_config:deps_info:owned_resources_dirs\) --resource-sources=@FileArg\(gen/components/version_info/android/version_constants_java.build_config:deps_info:owned_resources_zips\)
Traceback (most recent call last):
  File "../chromium/src/build/android/gyp/lint.py", line 400, in <module>
    sys.exit(main())
  File "../chromium/src/build/android/gyp/lint.py", line 396, in main
    add_pydeps=False)
  File "/home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium/src/build/android/gyp/util/build_utils.py", line 653, in CallAndWriteDepfileIfStale
    pass_changes=True)
  File "/home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium/src/build/android/gyp/util/md5_check.py", line 87, in CallAndRecordIfStale
    function(*args)
  File "/home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium/src/build/android/gyp/util/build_utils.py", line 638, in on_stale_md5
    function(*args)
  File "../chromium/src/build/android/gyp/lint.py", line 390, in <lambda>
    silent=args.silent),
  File "../chromium/src/build/android/gyp/lint.py", line 150, in _OnStaleMd5
    os.symlink(os.path.abspath(src), PathInDir(src_dir, src))
OSError: [Errno 17] File exists
[root@machine] #
```

This is the print output that appears right before `FAILED`:

```
src: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium-out/gen/components/version_info/android/java/org/chromium/components/version_info/VersionConstants.java
dst: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium-out/gen/components/version_info/android/java/org/chromium/components/version_info/VersionConstants.java
```

Other prints from invocations that did NOT fail:

```
src: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium/src/components/version_info/android/java/src/org/chromium/components/version_info/VersionConstantsBridge.java
dst: /tmp/tmp14icN6/SRC_ROOT1/components/version_info/android/java/src/org/chromium/components/version_info/VersionConstantsBridge.java
```

See?  The failing one is symlinking the file to itself.  The good ones symlink the file into `$TMPDIR`.

This is a **clean build**.  Everything under `~/chromium` was nuked before starting it.

Others have had this problem before:

https://groups.google.com/forum/#!topic/discuss-webrtc/yM8i7JuUkus

Rudd-O, 2019-03-13:
It's always the same file:

```
src: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium-out/gen/components/version_info/android/java/org/chromium/components/version_info/VersionConstants.java
dst: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/chromium-out/gen/components/version_info/android/java/org/chromium/components/version_info/VersionConstants.java
```

dan-v, 2019-03-13:
Thanks for the report @Rudd-O. I'll take a look as soon as I can.

dan-v, 2019-03-14:
I'm unable to reproduce this issue on v9.0.24 building 73.0.3683.75, so maybe something related to your local/jenkins changes.

closed 2019-03-14 by dan-v