dan-v/rattlesnakeos-stack #113: Fastboot/USB broken with PQ2A.190305.002

opthomas-prime, 2019-03-24:
Just wanted to inform you that some weird stuff is happening with the Pixel 2. I'm not sure if this is relevant for this project (Which his awesome BTW, Thanks!), but:
It seems like the boot-image/modem-image/fastboot is buggy right now. Yesterday i built a vanilla AOSP (PQ2A.190305.002) without any modifications - which kind-of bricked my phone (https://forum.xda-developers.com/pixel-2/help/pixel-2-bricked-t3914178). Until today i thought the problem as induced by incomplete binary/vendor-blobs from https://developers.google.com/android/drivers. But then i tried to flash the initial factory-image built with rattlesnake-stack and a similar problem happened again (https://gist.github.com/opthomas-prime/17489a49b3341c50c9f97236fc55aced). After flashing radio_a (+ reboot), fastboot were not able to find any device (This time dmesg showed some strange errors about how the cable could be faulty). When i tried to use fastboot some hours later on my desktop PC (Tried to flash on the notebook before) - everything worked fine. So well, everything is working now. Since multiple users are affected right now it would be more safe to wait until this problem is resolved on upstream before trying to apply updates/flash using USB/fastboot IMO.

dan-v, 2019-03-31:
Hi @opthomas-prime - was this issue resolved for you? Definitely seems like a bad cable or something like that.

opthomas-prime, 2019-04-01:
Hi @dan-v, Thanks for your answer. The issue is resolved for me. After using a non-Thunderbolt 3 port it simply worked. But i don't think this was about faulty cable. As you can see in https://gist.github.com/opthomas-prime/17489a49b3341c50c9f97236fc55aced, the port worked with the factory boot image, but after the triggered reboot, fastboot was not able to detect the phone. I haven't even touched the phone/cable in this time. This attracted my attention, since something similar happened to me when i flashed my own AOSP build (PQ2A.190305.002 - same as current rattlesnake-os). I just thought this may be some useful info...

dan-v, 2019-04-02:
Interesting - I'll keep this open and let's see if anyone else reports the issue. I haven't recently flashed as I'm always using the OTA updates, so I can't verify one way or the other at the moment.

9e3fce0a, 2019-12-14:
This happens to us all the time. It's almost always cable related. In VM environments we also have to make sure the guest has a USB 3.0 controller. Original Google-supplied cables always seem to work. Third party cables (either USB3-USBC or USBC-USBC) often give us problems.

dan-v, 2019-12-14:
@9e3fce0a - thanks for these details. I'm gonna close this issue out as I haven't heard any additional reports, and I think you are probably right that it may be cable related.

closed 2019-12-14 by dan-v