dan-v/rattlesnakeos-stack #114: AOSP branch config and fallback

pgera, 2019-04-01:
I'm hitting this error today "ERROR: Unable to get latest AOSP branch information. Stopping build. This can happen if https://source.android.com/setup/start/build-numbers hasn't been updated yet with newly released factory images." since the branch hasn't been updated. In cases like these, it should fall back to the last month's tag. Also, I don't see a config parameter to specify the build. Like chromium's version, it would be nice to have a config parameter for this too.

dan-v, 2019-04-02:
I chose to just bail out on the build when this occurs rather than guess what is wanted as this state generally exist for just a few hours before the build number is updated on AOSP site. If you think falling back is important I'm open to a PR for that.

I do think being able to specify the AOSP build number and branch could be exposed, so I'll repurpose this as feature request for that.

dan-v, 2019-09-16:
I don't expose this as a config option as it doesn't seem like something that should be static, but it is possible to specify one time during build start with --aosp-branch and --aosp-build

closed 2019-09-16 by dan-v