dan-v/rattlesnakeos-stack #116: Pixel 3a and 3a XL (`sargo` and `bonito`)

transitracer, 2019-05-09:
I know there's not been much time yet, but is there any idea whether rattlesnakeos-stack will be relatively trivial to adapt to the Pixel 3a line (/ what work will need to be done)

Plus, is there anything that could be done to expedite this process by anyone interested? This line of phones would and will definitely open up this privacy-conscious option to those who aren't as interested in spending as much money on a phone as the rest of the Pixel line requires.

dan-v, 2019-05-10:
Yes, I do plan on adding pixel 3a support. I don't plan on purchasing one, so it might be a little slow moving. It really depends on how quickly someone can add support for pixel 3a to android-prepare-vendor.

transitracer, 2019-05-10:
I have a 3a in the mail, so I'm willing to help with things requiring an actual device. Will poke into android-prepare-vendor!

PieDelivery, 2019-05-19:
See here: https://github.com/anestisb/android-prepare-vendor/issues/159#issuecomment-493734312

dan-v, 2019-05-24:
Support for Pixel 3a and 3a XL now exists thanks to @PieDelivery

closed 2019-05-24 by dan-v