dan-v/rattlesnakeos-stack #117: UnauthorizedOperation when calling the RequestSpotFleet operation

losuler, 2019-05-10:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options

### Description

Initially I received the error when the automated build ran, which also occurred afterwards when I attempted to manually start a build (see bottom for output).

This was odd to me because I've been running builds (automated and manually) for a few months now without issues and I haven't changed anything in AWS.

I'm not familiar enough with AWS to understand what permissions this error is involved with, but I checked the following things in case they help diagnose the problem:

- I am able to successfully launch a `c5.4xlarge` instance.
- The user with `AdministratorAccess` is still present in the management console.
- `aws s3 ls` command runs correctly and without errors.

### Platform
Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
chromium-version = ""
device = "walleye"
encrypted-keys = true
hosts-file = ""
ignore-version-checks = false
instance-regions = "ap-southeast-1"
instance-type = "c5.4xlarge"
max-price = "1.00"
region = "ap-southeast-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "rattlesnakeos"
```

### `./rattlesnakeos-stack build start`
```
INFO[0000] Using config file: /home/user/.rattlesnakeos.toml 
FATA[0006] Failed to start manual build. Function error: Unhandled. Output: {"errorMessage": "An error occurred (UnauthorizedOperation) when calling the RequestSpotFleet operation: You are not authorized to perform this operation.", "errorType": "ClientError", "stackTrace": [["/var/task/lambda_spot_function.py", 180, "lambda_handler", "response = client.request_spot_fleet(SpotFleetRequestConfig=spot_fleet_request_config)"], ["/var/runtime/botocore/client.py", 314, "_api_call", "return self._make_api_call(operation_name, kwargs)"], ["/var/runtime/botocore/client.py", 612, "_make_api_call", "raise error_class(parsed_response, operation_name)"]]}
```

dan-v, 2019-05-10:
@losuler - I am not positive on what would cause this out of the blue but there are two things we could check:
1) Could you post the following. Go to roles in IAM console https://console.aws.amazon.com/iam/home?/roles#/roles and click on
* `<stackname>-lambda` and then click on `<stackname>-lambda-policy`->JSON and paste that.
* `<stackname>-spot-fleet-role` and then click on `<stackname>-splot-fleet-policy`->JSON and paste that. 
2) Go to the Lambda console in the region where your stack is and click on your function `<stackname>-build`, scroll down to Execution role and just verify that it is using the `<stackname>-lambda-policy`

losuler, 2019-05-10:
## IAM Console
`<stackname>-lambda-policy`:
```json
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "ec2:DescribeSubnets",
                "ec2:RequestSpotFleet",
                "ec2:DescribeSpotPriceHistory",
                "iam:CreateServiceLinkedRole",
                "sts:GetCallerIdentity",
                "sns:Publish",
                "ec2:DescribeKeyPairs"
            ],
            "Resource": "*"
        }
    ]
}
```

`<stackname>-spot-fleet-policy`:
```json
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "ec2:DescribeImages",
                "ec2:DescribeSubnets",
                "ec2:RequestSpotInstances",
                "iam:CreateServiceLinkedRole",
                "ec2:TerminateInstances",
                "ec2:DescribeInstanceStatus",
                "iam:PassRole"
            ],
            "Resource": [
                "*"
            ]
        },
        {
            "Effect": "Allow",
            "Action": [
                "elasticloadbalancing:RegisterInstancesWithLoadBalancer"
            ],
            "Resource": [
                "arn:aws:elasticloadbalancing:*:*:loadbalancer/*"
            ]
        },
        {
            "Effect": "Allow",
            "Action": [
                "elasticloadbalancing:RegisterTargets"
            ],
            "Resource": [
                "*"
            ]
        }
    ]
}
```
## Lambda Console
I can confirm that it's using the `<stackname>-lambda` role for the execution role.

dan-v, 2019-05-10:
Hmm that all looks fine. Could you try to start a build through the Lambda console by going to the function and clicking 'Test'.

losuler, 2019-05-10:
It's returning the same error I pasted in my original post (the json following `FATA[0006] Failed to start manual build. Function error: Unhandled. Output:`).

This is the log output (I redacted the ID codes in case they were sensitive):
```
START RequestId: <redacted> Version: $LATEST
ap-southeast-1c 0.255400
ap-southeast-1a 0.268300
ap-southeast-1b 0.254000
Requesting spot instance in AZ ap-southeast-1b with current price of 0.254000
Sent SNS message There was a problem requesting a spot instance c5.4xlarge: An error occurred (UnauthorizedOperation) when calling the RequestSpotFleet operation: You are not authorized to perform this operation. and got response: {'MessageId': '<redacted>', 'ResponseMetadata': {'RequestId': '<redacted>', 'HTTPStatusCode': 200, 'HTTPHeaders': {'x-amzn-requestid': '<redacted>', 'content-type': 'text/xml', 'content-length': '294', 'date': 'Fri, 10 May 2019 06:42:49 GMT'}, 'RetryAttempts': 0}}
An error occurred (UnauthorizedOperation) when calling the RequestSpotFleet operation: You are not authorized to perform this operation.: ClientError
Traceback (most recent call last):
  File "/var/task/lambda_spot_function.py", line 180, in lambda_handler
    response = client.request_spot_fleet(SpotFleetRequestConfig=spot_fleet_request_config)
  File "/var/runtime/botocore/client.py", line 314, in _api_call
    return self._make_api_call(operation_name, kwargs)
  File "/var/runtime/botocore/client.py", line 612, in _make_api_call
    raise error_class(parsed_response, operation_name)
botocore.exceptions.ClientError: An error occurred (UnauthorizedOperation) when calling the RequestSpotFleet operation: You are not authorized to perform this operation.

END RequestId: <redacted>
REPORT RequestId: <redacted>	Duration: 4649.39 ms	Billed Duration: 4700 ms 	Memory Size: 128 MB	Max Memory Used: 75 MB	
```

dan-v, 2019-05-10:
That is really strange. I am only able to reproduce this issue if I take RequestSpotFleet out of the `<stackname>-lambda` IAM role. Just to see if it works at all, could you go to the `<stackname>-lambda` IAM role and attach the 'AdministratorAccess' policy. Then try kicking off the Lambda function again.

losuler, 2019-05-10:
Yeah, it succeeded once I attached the `AdministratorAccess` policy.

dan-v, 2019-05-10:
Be aware that when you 'deploy' your stack again, the underlying terraform code will likely try to remove `AdministratorAccess` from this policy. I'm curious if you manually remove `AdministratorAccess` from it now after being able to build if it will still fail. Still a bit of a mystery on what is going on here unfortunately.

losuler, 2019-05-11:
It failed again with the same error as before once I removed `AdministratorAccess`.

losuler, 2019-05-11:
I edited the json for the `<stackname>-lambda-policy` so it allowed all actions (`"Action": "*",`) and it succeeded. So I'm guessing the policy is just missing a permission then?

dan-v, 2019-05-18:
Still unsure what permission might be missing in your scenario. I am not able to reproduce with a matching IAM policy. I'll leave this open for now in case others bump into it and maybe we can track down further.

losuler, 2019-05-24:
After spending some time trying to figure this out, I finally found the cause of this. Turns out it was the region `ap-southeast-1` that was causing the error. For as soon as I removed it and added a new region (`ap-northeast-1`), it ran successfully.

This seems really odd, perhaps it's an issue that AWS need to fix? Is there a way to report this to them?

johan-adriaans, 2019-05-29:
Odd, I have the same problem in eu-central-1, eu-west-1 and eu-west-3. I have been trying some extra SpotFleet permissions but no luck so far.

dan-v, 2019-05-29:
@johan-adriaans and @losuler - thanks for this additional information. Still not sure why this doesn't work in certain regions with the same permissions, but definitely helpful in tracking down what is going on. If either of you can come up with something better than enabling admin permissions (as i don't want to do this) as a workaround please let me know.

agates, 2019-05-29:
The visual policy editor is giving these warnings.

[image]

dan-v, 2019-05-29:
Thanks for chiming in @agates. The `<stackname>-spot-fleet-role` is essentially just a duplicate of the default AmazonEC2SpotFleetTaggingRole role, which has these same warnings in it. The issue here appears to be related to the `<stackname>-lambda` policy not having permissions to call RequestSpotFleet in certain regions. I should have enough details on this to try to track it down when I have some free time. Enabling cloudtrail may give some additional hints as to what is going on here.

agates, 2019-05-30:
Hmm.  Yeah I think I thought something I changed in there fixed the issue, but in hindsight I think that was a false positive -- changes to IAM policies don't appear to update as quickly as I can test.

mmahfoudi, 2019-05-30:
I have the same error...

dan-v, 2019-05-30:
I was able to reproduce issue in ap-southeast-1. I added 'iam:PassRole' to the inline policy defined in role `<stackname>-lambda` and that seemed to do the trick. If someone could confirm that works for them or not that would be helpful.
```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "ec2:DescribeSubnets",
                "ec2:DescribeSpotPriceHistory",
                "ec2:RequestSpotFleet",
                "iam:CreateServiceLinkedRole",
                "iam:PassRole",
                "sts:GetCallerIdentity",
                "sns:Publish",
                "ec2:DescribeKeyPairs"
            ],
            "Resource": "*"
        }
    ]
}
```

johan-adriaans, 2019-05-30:
Awesome, i'm not near a computer today. I will try tomorrow and let you know.

dan-v, 2019-05-30:
Thanks @johan-adriaans!

losuler, 2019-05-30:
@dan-v Did it get to the build stage after you added `iam:PassRole`? Because when I was trying to figure out the issue last week I tried the same thing except the spot fleet request would never be fulfilled (i.e. I would reach the "Successfully requested a spot instance." stage but not the "RattlesnakeOS Build STARTED" stage), thus forcing me to manually end the request (I waited 30 minutes on 4 different tries).

At the time I tried increasing the `max-price` and `skip-price` despite the error not indicating it was an issue with the price, and that didn't solve it.

If it is getting to the build stage, I'll try adding the role again and confirm it's working for me too.

dan-v, 2019-05-30:
@losuler - it did not make it to building only because EC2 doesn't have capacity in ap-southeast-1 for c5.4xlarge instances. That unfortunately I don't have control over. I'd definitely recommend doing builds in the default regions defined if possible as they generally have the most capacity and the cheapest prices.

dan-v, 2019-05-30:
I pushed a new version with the iam:PassRole fix: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v9.0.28

mmahfoudi, 2019-05-30:
It works for eu-west-3 region !

dan-v, 2019-05-30:
thanks for update @mmahfoudi!

johan-adriaans, 2019-05-31:
Yep, works for me! Thanks ^^

By the way, how did you figure out you needed this permission? I take it you didn't just try them one by one. I'm not well versed in the ways of AWS and bugs like this make me feel like a blind man in a china shop :)

RattlesnakeOS Build STARTED
  Device: taimen
  Stack Name: rattlesnakeos-taimen-xxxxxxx
  Stack Version: 9.0.28
  Stack Region: eu-central-1
  Release Channel: taimen-stable
  Instance Type: c4.4xlarge
  Instance Region: eu-central-1
  Instance IP: x.x.x.x
  Build Date: 2019.05.31.08
  Elapsed Time: 0hrs 0min 13sec
  AOSP Build: PQ3A.190505.001
  AOSP Branch: android-9.0.0_r36
  Chromium Version: 74.0.3729.157
  F-Droid Version: 1.6.2
  F-Droid Priv Extension Version: 0.2.9
  Build Reason: 'Stack version 9.0.25 != 9.0.28' 'Chromium version 74.0.3729.136 != 74.0.3729.157' 'F-Droid version 1.6 != 1.6.2'

agates, 2019-05-31:
9.0.28 fixes it for me as well.

> RattlesnakeOS Build STARTED
>   Device: walleye
>   Stack Name: redacted
>   Stack Version: 9.0.28
>   Stack Region: ca-central-1
>   Release Channel: walleye-stable
>   Instance Type: c5.4xlarge
>   Instance Region: us-east-2
>   Instance IP: redacted
>   Build Date: 2019.05.31.16
>   Elapsed Time: 0hrs 1min 10sec
>   AOSP Build: PQ3A.190505.001
>   AOSP Branch: android-9.0.0_r36
>   Chromium Version: 74.0.3729.157
>   F-Droid Version: 1.6.2
>   F-Droid Priv Extension Version: 0.2.9
>   Build Reason: 'Stack version 9.0.27 != 9.0.28'

dan-v, 2019-05-31:
Thanks all!

@johan-adriaans - to track this down I used a combination of CloudTrail to see what API call were attempting to be made under the covers when executing the Lambda function and just a simple comparison to the default Lambda role they will auto generate for you.

closed 2019-05-31 by dan-v