dan-v/rattlesnakeos-stack #121: Builds timeout due to configuration prompt

PieDelivery, 2019-06-14:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [ ] I am able to reproduce this issue without any advanced customization options

### Description

[Description of the issue]

### Steps to Reproduce

1. Start a build
2. Wait

**Expected behavior:** It builds

**Actual behavior:** The build successfully starts, but never completes.

### Platform
Windows/Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-server= false
chromium-version = ""
device = "sargo"
encrypted-keys = false
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "<not important>"
region = "us-east-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.00"
ssh-key = "rattlesnakeos"
```
I was testing this with microg so that's enabled as well (my instance allotment isn't high enough to perform another test without it right now).

### Carrier
N/A

### Email Notification Output
There is a notification for successfully starting the build, but nothing beyond that.

### Full log
There is no log file. I've attached a screenshot from tail -f /var/log/cloud-init-output.log.
[image]

dan-v, 2019-06-14:
Oh man, sorry to hear that @PieDelivery! I think just adding `export DEBIAN_FRONTEND=noninteractive` near top of build script will prevent this issue. If you have time to test that fix and/or do a PR it would be much appreciated. I will take a look as soon as I can otherwise.

PieDelivery, 2019-06-14:
I just found a stack overflow thread saying the same thing, so I'll give it a whirl and see what happens.

PieDelivery, 2019-06-14:
Should be fixed with https://github.com/dan-v/rattlesnakeos-stack/pull/122.

closed 2019-06-14 by PieDelivery