dan-v/rattlesnakeos-stack #124: Attestation instance terminated

mmahfoudi, 2019-06-25:
### Description

Hello yall,
I started a build with attestation server, it appears that the elastic IP is not attached to my attestation server.
The health of the app is GREEN but no instance is running. Before the instance was terminated I was able to access to the URL but only HTTP and not HTTPS.

How can I solve this issue please ?

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-instance-type = "t3.nano"
attestation-max-price = ".01"
attestation-server = true
chromium-version = ""
device = "blueline"
encrypted-keys = true
hosts-file = "https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts"
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.9xlarge"
max-price = "2.00"
region = "eu-west-3"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "rattlesnakeos"
```

### Email Notification Output
```
RattlesnakeOS Build SUCCESS
 Device: blueline
 Stack Name: rattlesnakeos-blueline-xxx
 Stack Version: 9.0.30 
 Stack Region: eu-west-3
 Release Channel: blueline-stable
 Instance Type: c5.9xlarge
 Instance Region: us-east-2
 Instance IP: 18.225.32.214
 Build Date: 2019.06.24.23
 Elapsed Time: 3hrs 16min 59sec
 AOSP Build: PQ3A.190605.003
 AOSP Branch: android-9.0.0_r40
 Chromium Version: 75.0.3770.101
 F-Droid Version: 1.6.2
 F-Droid Priv Extension Version: 0.2.9
 Build Reason: 'Stack version 9.0.29 != 9.0.30' 'Chromium version 75.0.3770.67 != 75.0.3770.101'
```
```
Timestamp: Tue Jun 25 02:11:40 UTC 2019
Message: Removed instance 'i-03f993f30b097ad65' from your environment.  (Reason: Instance is in 'terminated' state)

Environment: attestation
Application: rattlesnakeos-blueline-xxx-attestation

Environment URL: http://attestation.xxx.eu-west-3.elasticbeanstalk.com
```
```
Timestamp: Tue Jun 25 02:10:47 UTC 2019
Message: New application version was deployed to running EC2 instances.

Environment: attestation
Application: rattlesnakeos-blueline-xxx-attestation

Environment URL: http://attestation.xxx.eu-west-3.elasticbeanstalk.com
```

dan-v, 2019-06-25:
Hi @mmahfoudi. This is a pretty new feature, and I don't know how many people are using it yet (other than me), so I expect there will be some bugs. I don't have any sort of writeup on best way to debug this yet, so I'll just give you a few off the top of my head ideas for debugging.

If you go look at the status of your elasticbeanstalk app in the console, what do you see (https://eu-west-3.console.aws.amazon.com/elasticbeanstalk/home?region=eu-west-3#/applications). Is it red? If you click on 'attestion' app what do the recent events show? If you click on 'Configuration' what does it show under 'Instances'.
* If there is an instance, you could grab logs, go to Logs->Request Logs->Full and that should download a zip file. I don't have a good mechanism here for uploading logs, so if you want to shoot me an email dan@vittegleo.com with that, I can take a look.
* If there is no instance, it's possible there isn't capacity in this eu-west-3 region for this instance type. One thing worth trying might be to in the console click Action->Rebuild environment.

mmahfoudi, 2019-06-25:
Yeah this is relatively new, I thought other users would use it.
Anyway, the environment was still green, tried to rebuild the environment but could not start an instance.
I relaunched a build without the feature and I will wait for more maturity.
Can we keep this issue opened ?

dan-v, 2019-06-25:
Unfortunately there's nothing actionable at this point, so I'm not sure if leaving it open will help. 

My best guess based in information provided is that this region doesn't have enough capacity for t3.nano instances. To verify this, can you go to https://eu-west-3.console.aws.amazon.com/ec2/autoscaling/, click on the autoscaling group there (awseb-*), click on 'Activity History' tab and see why the instance is not launching. 

If it is a capacity issue, there are a few options that I can think of to help improve the situation:
* Could add an option to use an on demand instance to ensure capacity, but this of course costs quite a lot more per month.
* Allow selecting a different region than the primary stack so that you could choose a region with larger instance capacity (e.g. us-east-2, us-east-1, us-west-2, etc).

dan-v, 2019-06-27:
Gonna close this one out unless we can get more details. I've opened a new issue for possible ways to improve instance availability https://github.com/dan-v/rattlesnakeos-stack/issues/125.

closed 2019-06-27 by dan-v