dan-v/rattlesnakeos-stack #129: Allow custom build manifests

dan-v, 2019-07-19:
Allow entire build manifest to be swapped out or provide a way to easily edit/replace lines so that things like customized memory allocator, kernel build/customizations, etc.

ubergeek77, 2019-07-23:
Out of curiosity, would this feature allow someone to build RattlesnakeOS for use in an Android Virtual Device/QEMU-based emulator? Be it arm64 or x86 (though preferably arm64 for build parity).

I only have a single Pixel phone - my Pixel 3 XL - and it is my daily driver. Unfortunately, I don't have access to another. As much as I would like to go all in on RattlesnakeOS, I really need a way to test a built image in an emulator of some sort just to make sure I won't brick my phone or bootloop upon first installation.

dan-v, 2019-07-23:
Hi @ubergeek77. This request would likely fall under the issue of creating a better development environment (https://github.com/dan-v/rattlesnakeos-stack/issues/86). I don't know of the feasibility of building an image that will run in an emulator though as these images are specific to the device they are being built for in terms of drivers, kernel, firmware, etc. But I agree that would be useful if possible.

You should be fine as long as you don't lock your bootloader while testing. If the image you created was broken in some way and it doesn't boot or bootloops you can always get into recovery mode and load stock images from Google.

ubergeek77, 2019-07-23:
Thanks for the insight! In that case, I'll have to pick a day where I don't think I'll be texted/called for a few hours and see if I can set this up.

I was also concerned about potentially making a bad OTA update due to a custom patch "gone wrong," since at that point my bootloader would be locked. But I just realized as I was typing this comment that the Pixel 3 XL uses A/B partitions. So I'm assuming one bad OTA would put me back to the previously-working partition for me to address the issue later (though I would probably have to remember to disable auto-updates, I've never encountered a bad OTA before so I'm not sure how this is handled).

dan-v, 2019-07-23:
Yeah, a bad OTA update will recover after a few failed boot attempts thanks to the A/B partitioning. And in that case you're right you would want to disable the updater so it didn't keep downloading and installing a bad update.

davze, 2020-11-29:
Is it possible to include a custom kernel repo atm?

dan-v, 2020-11-29:
> Is it possible to include a custom kernel repo atm?

Not currently. There was some work contributed to enable this in the 10.0 branch, but it was never fully tested or turned on and I ripped it out for 11.x. I'd be happy to take a PR that enables custom kernel building though.

PR #182 (2021-03-01): refactoring and changes to how customizations work
* new: add support for custom config repos and deprecate old customization methods (custom-patches, custom-manifest-remotes, custom-manifest-projects, custom-prebuilts)
* new: also add support for core config repos that can be used to build an OS with plain AOSP rather than using the default RattlesnakeOS core. closes #129.
* new: add support for Pixel 4a 5G. closes #180.
* deprecate: encrypted-keys option (for now) to reduce build script complexity
* deprecate: hosts-file (can be added in custom config repo)
* move version checking to lambda  [... cut]
commit 61be4195 (2021-03-01): refactoring and changes to how customizations work (#182)

* new: add support for custom config repos and deprecate old customization methods (custom-patches, custom-manifest-remotes, custom-manifest-projects, custom-prebuilts)
* new: also add support for core config repos that can be used to build an OS with plain AOSP rather than using the default RattlesnakeOS core. closes #129.
* new: add support for Pixel 4a 5G. closes #180.
* deprecate: encrypted-keys option (for now) to reduce build script complexity
* deprecate: hosts-file (can be added in custom config repo)
* move version checking to [... cut]
closed 2021-03-01 by dan-v