dan-v/rattlesnakeos-stack #135: Support Aurora Services (privileged extension for Aurora Store)

articlecat, 2019-09-26:
**Is your feature request related to a problem? Please describe.**
Since RattlesnakeOS has no root access, Aurora Store requires the user to manually install or update every app downloaded from the store.

**Describe the solution you'd like**
Integrating [Aurora Services](https://gitlab.com/AuroraOSS/AuroraServices) (a fork of the [F-Droid privileged extension](https://gitlab.com/fdroid/privileged-extension) for Aurora Store) would allow Aurora Store to install and update apps in the background. RattlesnakeOS already supports the F-Droid privileged extension.

**Describe alternatives you've considered**
The main alternative would be to support root access in RattlesnakeOS, but I understand that the developers don't want to introduce this feature into RattlesnakeOS. Another alternative would be to use LineageOS or another operating system that supports root access.

**Additional context**
[Aurora Store](https://gitlab.com/AuroraOSS/AuroraStore) is a free and open source client for Google Play Store. It was originally forked from [Yalp Store](https://github.com/yeriomin/YalpStore), but was later rewritten.

dan-v, 2019-10-03:
I'd like to avoid adding additional software at this point (I'd even like F-Droid to be an opt in), as I would like this OS to essentially be barebones AOSP by default and then you can opt into adding additional software and changes. The current mechanisms for customizing is not where i'd like it to be in terms of ease of use, but I'm hoping to improve that in the future. This is definitely possible to do today using the shell script option, but would require a bit of technical knowledge of how to swap out the default keys, compile the app, and add to the AOSP build.

I'm gonna leave this request open for now so it stays on my mind, as it's something I do want to support, but it's just not something I want by default baked into the OS.

mgidding, 2019-10-04:
I also wanted this functionality so I created a repo for it:
https://gitlab.com/mgidding/aurora-store

You can use the repo by adding the following snippet to your rattlesnakeos-stack config:
`[[custom-prebuilts]]
modules = ["AuroraStore", "AuroraServices"]
repo = "https://gitlab.com/mgidding/aurora-store"`

If you don't trust the apks in my repo you can fetch them yourself from https://gitlab.com/AuroraOSS/AuroraStore/-/tags.

dan-v, 2021-03-01:
Not going to add this as part of core.

closed 2021-03-01 by dan-v