dan-v/rattlesnakeos-stack #140: Merging upstream changes from the Updater app

scott-8, 2019-11-27:
I was looking into messing around with the Updater app. I noticed that the Rattlesnake Updater app is pretty behind the GrapheneOS one. Before I pull in those changes, I wanted to see if you had tested those them at all or if any of the changes had a potential conflict with RattlesnakeOS that you knew about.

dan-v, 2019-11-29:
Hi @scott-8! Sorry for the delay in response. I haven't tracked closely the changes that were contributed from CalyxOS updater into GrapheneOS, but briefly looking at them it does look like some very useful changes that give much better visibility into the updating process. I don't think there are any breaking changes from what I see. I'd be super happy if you could pull these changes in and test it out and happy to take a PR for that.

scott-8, 2019-12-03:
No worries, I've been busy too. I test it out soon and let you know how it goes.

scott-8, 2019-12-08:
Everything runs fine with the new updater changes. It successfully pulls pulls updates from S3 and installs them. I didn't really change anything from upstream, just the URL and app name (https://github.com/scott-8/platform_packages_apps_Updater). The current repo doesn't have the same commit history so I made a PR on the dev branch of your Updater repo.

scott-8, 2019-12-09:
Also, I'm adding two new features to the updater app: ability to change the server URL and how often it checks for updates. If you want me to submit a PR for either one, let me know (not sure if you feel like they're needed, I added them for serving the update from a local server). I've only done the server URL one so far.

dan-v, 2019-12-09:
@scott-8 thanks! I'll pull it into dev branch and test it out. If all looks good, I'll just rewrite history on 10.0 branch so it's easier to pull in changes in the future.

I would definitely pull in the change for how often it checks for updates, but not sure on the changing the server URL. The only reason I say that is there may be some security implications with giving the ability to change this URL. I feel like the likelihood of this happening is quite low, but if an attacker could change this to an invalid location they could prevent a phone from getting updates or if an attacker had your signing keys they could point it at a malicious update server.

scott-8, 2019-12-09:
I'll let you know once I've tested it out. Looking at your point on the URL, how do you feel about "never" being one of the options for update frequency (i.e. turn off updates)? I was going to add it as an option for people who only want to check for updates manually, but it could potentially lead to poorer security.

That's a good point about the security implications of opening the URL to changes. Plus, I don't think most people would find it useful. With stack deployed, there would never be a reason to change it.

dan-v, 2019-12-09:
I do feel like being able to set never can be useful. For me it would be mostly useful in fringe situation where let's say I build a bad update and want to stop it from continuously downloading. I guess today in the updater app you can already swap over to a different channel (e.g. beta) which in theory could do the same thing of preventing updates. And with that in mind, I think I'd be open to "never" being an option.

dan-v, 2019-12-09:
I verified all works as expected and pulled into 10.0 branch. Thanks @scott-8! Gonna close this out, but happy to take more PRs :)

closed 2019-12-09 by dan-v