dan-v/rattlesnakeos-stack #142: Issues with F-droid privileged extension on 10.0

dan-v, 2019-12-07:
Need to track down what is going on here. Some apps install fine with no prompts while others will prompt asking to enabled installing unknown apps (which shouldn't be happening).

pgera, 2019-12-07:
```
adb logcat | grep PrivilegedExtension
12-07 07:19:44.710  1282  1282 I PackageManager: /system/priv-app/F-DroidPrivilegedExtension changed; collecting certs
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: Android not compatible!
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: java.lang.NoSuchMethodException: android.app.ApplicationPackageManager.installPackage [class android.net.Uri, interface android.content.pm.IPackageInstallObserver, int, class java.lang.String]
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at java.lang.Class.getMethod(Class.java:2072)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at java.lang.Class.getMethod(Class.java:1693)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at org.fdroid.fdroid.privileged.PrivilegedService.onCreate(PrivilegedService.java:272)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.handleCreateService(ActivityThread.java:3953)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.access$1500(ActivityThread.java:219)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1875)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.os.Handler.dispatchMessage(Handler.java:107)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.os.Looper.loop(Looper.java:214)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at java.lang.reflect.Method.invoke(Native Method)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
12-07 07:19:55.594  4787  4787 E PrivilegedExtension: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: Android not compatible!
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: java.lang.NoSuchMethodException: android.app.ApplicationPackageManager.installPackage [class android.net.Uri, interface android.content.pm.IPackageInstallObserver, int, class java.lang.String]
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at java.lang.Class.getMethod(Class.java:2072)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at java.lang.Class.getMethod(Class.java:1693)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at org.fdroid.fdroid.privileged.PrivilegedService.onCreate(PrivilegedService.java:272)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.handleCreateService(ActivityThread.java:3953)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.access$1500(ActivityThread.java:219)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1875)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.os.Handler.dispatchMessage(Handler.java:107)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.os.Looper.loop(Looper.java:214)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at java.lang.reflect.Method.invoke(Native Method)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
12-07 07:23:44.101  4787  4787 E PrivilegedExtension: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)

```

dan-v, 2019-12-09:
There is also an issue with F-droid privileged extension version parsing that was preventing it from updating to the latest. Even on 1.8 alpha client and 0.2.11 privileged extension, I'm still seeing some issues. For any apps targeting SDK 29 the normal workflow of installing apps in F-Droid works as expected with no prompts for unknown sources, but for any apps targeting older SDKs, it prompts to enable installing unknown sources.

chirayudesai, 2019-12-09:
This should be fixed with https://gitlab.com/fdroid/fdroidclient/merge_requests/856

For future reference, if there are still issues, knowing which app in particular triggers it helps a ton in reproducing it and then fixing it. This particular one was easily reproduced by "Simple Contacts", thanks @dan-v

closed 2020-09-25 by dan-v