dan-v/rattlesnakeos-stack #144: Updater is downloading and installing OTA image multiple times, although the verification failed

Rot127, 2019-12-09:
### Description

During the deployment process my keys got mixed up somehow and the OTA image where build (successfully) with a different key pair. When the updater downloaded the OTA image it wasn't able to verify and install it. No warning message was given. Instead it downloaded the image again and again causing huge costs on my Amazon account for using too much traffic.

The Rattlesnake Stack version was 9.0.36.

**Expected behavior:** When the install/download of an update fails the updater should show a warning message and wait for an confirmation of the user before doing something else.
Additionally, the FAQ should list this scenario under its __cost section__.

**Actual behavior:** The update repeatably downloaded and tried to install the OTA image without warning.

### Platform
Linux

scott-8, 2019-12-09:
It would be pretty easy to do this, I could probably make a notification for a failed update, but I'm not familiar enough with notifications to add actions to it.

For anyone that wants to try, I think all you would have to do is replace [this line](https://github.com/RattlesnakeOS/platform_packages_apps_Updater/blob/10.0/src/app/seamlessupdate/client/Service.java#L321) with something that creates a notification with button that performs that action instead (and a cancel button that dismisses it).

dan-v, 2019-12-09:
@Rot127 - sorry to hear about that; it's a pretty bad edge case for sure. On 10.0 this isn't prevented, but @scott-8 pulled in latest upstream downloader now which shows when an OTA is downloading with progress bar, so it would at least be a lot more evident something wrong was happening. I'd love to have your logs from the successful build that incorrectly regenerated your keys as that definitely shouldn't ever happen. If there was some issue talking to S3 when looking for keys, I think that could cause this to happen, so I'll make fixing that scenario the action item of this issue. I don't think it makes sense to add a specific warning for this edge case to the README and instead just work on fixing the core issue. I think a blanket warning about costs in face of software bugs could make sense though.

Rot127, 2019-12-11:
@dan-v Unfortunately I deleted all s3 buckets and forgot to backup the logs, sorry. But the issue was probably triggered by two build configurations simultaneously running on AWS. I deployed a second one for testing purposes.
Though I have to correct my assumption that the bug was a signing key issue. I knew that the OTA image verification failed because of the Logcat logs I could obtain because my phone is rooted.
I assumed that it bug occurred because of mixed signing keys because the building process was successful. Which can be obviously wrong.
So sorry again for not backing up the logs.

dan-v, 2019-12-23:
@Rot127 - no worries. I think even if your assumption about the signing key issue wasn't true, it may certainly still be possible for that scenario to play out. The updater change that @scott-8 mentioned has been merged so hopefully that would help make this issue much more apparent. I'm going to leave this open until I can at least verify the behavior in failure cases when checking for signing keys in S3 as I think this deserves a closer look.

dan-v, 2020-02-06:
I'm gonna close this one out. The new updater makes it much more obvious that downloads/errors are happening and also has the option to disable if it were in a constant bad state.

closed 2020-02-06 by dan-v