dan-v/rattlesnakeos-stack #145: Networkstack keys not copied to S3 bucket when using plaintext keys

9e3fce0a, 2019-12-14:
### Prerequisites

* [X] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [X] I am able to reproduce this issue without any advanced customization options

### Description

Networkstack keys not copied to S3 bucket when using plaintext keys

### Steps to Reproduce

1. Deploy and run a new build without encrypting keys
2. Examine the stackname-keys S3 bucket, noting that networkstack keys won't be there.

**Expected behavior:** networkstack keys should be sync'd to the keys bucket

**Actual behavior:** networkstack keys were not sync'd to the keys bucket

### Platform
e.g. Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
chromium-version = ""
device = "blueline"
encrypted-keys = false
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "<not important>"
region = "us-west-2"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.00"
ssh-key = "rattlesnakeos"
```

Bug is here: https://github.com/dan-v/rattlesnakeos-stack/blob/10.0/templates/build_template.go#L1201-L1202

dan-v, 2019-12-14:
Thanks for the report @9e3fce0a and for tracking down the issue. I'll make sure a fix goes in for this in next release.

dan-v, 2020-02-03:
Should be resolved in [v10.0.2 ](https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v10.0.2)

closed 2020-02-03 by dan-v