dan-v/rattlesnakeos-stack #146: Investigate integration with aosp-build

dan-v, 2019-12-23:
https://github.com/hashbang/aosp-build - This would provide a better build/development environment in my opinion. It would also be beneficial to share a common build system with another project as it would allow for much easier collaboration on fixes, tweaks, etc. 

General idea is:
* Keep much of the existing process of a launching a build instance
* Pull down slightly customized version of aosp-build for RattlesnakeOS
* Do some of the existing interaction with AWS specific things like S3 buckets to check versions, pull down keys, etc
* Remove majority of the build script as it exists today and replace with make targets in aosp-build. Essentially just do a local build with aosp-build (it just happens to be on AWS spot instance).
* Do more of the existing interaction with AWS specific things like uploading builds, bumping updater metadata, etc.
* The existing customization options might need to be tweaked or changed to support aosp-build.

ubergeek77, 2019-12-24:
I'm currently doing a full local build with aosp-build, and it's not done yet, but I'd really like to point out that it's pulling down the r10 tag for crosshatch instead of r15 like Rattlesnake does.

I'll look more into why this happens when I'm home from the holidays (currently dispatching all of this remotely to my home server from my phone), but this may be a potential pain point if aosp-build is always 1 month behind. It could also be a bug in aosp-build, so I think this needs more investigation for sure.

dan-v, 2019-12-24:
@ubergeek77 - I added a mechanism to aosp-build (which should exist in master) to update the config file with latest branches by doing `make DEVICE=crosshatch config manifest`. (config target is really the only thing that is pulling in latest branches into config file). If that isn't working, let me know and I'll try to track it down.

ubergeek77, 2019-12-24:
Ah, ok. I only did `make DEVICE=crosshatch` since I didn't see this mentioned in the Readme. In that case, aosp-build is fine in all likelihood. I just didn't start it correctly.

ubergeek77, 2019-12-24:
Thanks, that's what I had to do. I didn't realize the repo had a default config from October, and that there was a make target specifically for updating it (perhaps RattlesnakeOS has spoiled me a bit in that regard).

Once the build is finished and I have some time to play around with it, I'm interested in seeing how easy it is to migrate some of what you've got in your build template over to a local build with aosp-build. It's seeming like the vast majority (if not the entirety) of the non-build-related behavior in your build script is patch application (manually or user-supplied), user-supplied custom scripts, and lots of AWS stuff (logging, uploading, key generation, etc).

I'll leave the AWS stuff for you, but from a *very* high level, at the risk of oversimplifying things, I think you would "only" need to convert the manual patching your build.sh does into traditional patches, and a good chunk of the integration (in the context of your build script, at least) would be done. Of course, aosp-build's default build script would need to be modified slightly to make those mandatory and distinct from user patches, but other than that, the only major component I see needing to be integrated would be your concept of custom user scripts. Once again, those could just be integrated into aosp-build's default build script, or even a dedicated make target.

I really do hope I'm not downplaying the level of effort this would require, as that's not my intent. But I definitely think the functionality of aosp-build very much suits the type of project that RattlesnakeOS is.

dan-v, 2019-12-25:
Yeah, I don't think there's a ton of complexity involved in ripping out the existing build script and dropping in aosp-build as most of it is just the same build process just cleaned up/improved, but I think it will still take some time and effort to get all the ducks in a row. I think any of the base changes in the build script that deviate from aosp-build will become part of a separate rattlesnakeos repo just like https://github.com/hashbang/os has implemented it; the build process then uses the aosp-build docker image. I do still need to decide what I want to do with the custom patches, prebuilts, scripts, etc. Ideally each of those would be implemented in a way that nicely integrate with the aosp-build process. For example, prebuilts could just get appended into the config.xml file and patches could just get dumped into patches directory etc. Or maybe it would be better to just drop all of that complexity and just allow an entire aosp-build based repo to be specified (if you don't want to default rattlesnakeos one) that includes all of their desired patches/changes included. Still thinking about it a bit, but if you have any ideas on that front I'd love to hear them.

ubergeek77, 2019-12-25:
As a ~~lazy~~ end user, what I really like about the current RattlesnakeOS-stack is that builds are uniform and predictable. That is to say, I can specify certain patches, scripts, and prebuilts into `.rattlesnakeos.toml`, those will be added in a predictable order, and the rest of the build process is taken care for me. I'm not touching any of the build script, so as long as I don't do anything really stupid with my patches, I can basically guarantee a successful build every time I start a build.

Personally, for a migration to aosp-build like this, I would love it if I could continue to only have to maintain my own custom patch/script/prebuilt repositories, and still have those included as part of the build. This way, I don't have to maintain (or monitor the maintenance of) an entirely separate aosp-build fork, I don't have to worry about touching the main build script, I get to keep my current patch and script workflow, and those still get integrated into the build process in an equally predictable way as they currently are.

Why do I like this so much?

Given the complexity of AOSP in terms of CPU requirements to build, at-home internet speed to download, local storage to save, and sheer number of files, hacking away at parts of the build process trying to customize things can become very cumbersome very fast. Simple patches and prebuilts are one thing, but it can be difficult for someone with limited local hardware to test things such as build order, build commands, and etc. So, while the aosp-build repository is tiny, there's a lot for me to break there, and if the build process has to change for any reason (bug fixes, Android 11 is released, etc), I have to make sure my repository still makes functional builds. I could likely copy yours if you provide one, but I would still have to do that manually.

With a predictable, modular-but-maintained build process like we currently have, I have the advantage of highly knowledgeable people such as yourself to handle the build-centric tasks, and all I need to worry about are my customizations.

From my perspective at least, I see this as less complex when compared to maintaining a fork of aosp-build, like you mentioned.

As far as integration with RattlesnakeOS-stack goes, I imagine it would look like this:

- `rattlesnakeos-stack deploy` still lets the user define the patches, scripts, and prebuilts they want. These would go into aosp-build's config.yml. From the perspective of a maintainer (like you), this could be further simplified by requiring the user to specify a manifest file in their remote patch/script/prebuilt repository that replaces the role of .toml arrays, and instructs the script which things to include. Then, all that would need to be included in config.yml is a link to said repository.
- I *really* think current "hacks" in build_template.go could be converted into .patch files and custom scripts. They look relatively simple, so unless I'm missing something obvious (and I'm known for that), those could be made into patches and scripts with little friction.
- If that ends up working, then *every* patch, script, or prebuilt currently employed by RattlesnakeOS-stack could be converted into repositories of said items, and added into config.yml so that they are included in the build. I'm *pretty* sure yaml parsers understand key order, so the build process would consider whatever is defined first in whatever section to be run first. I notice that your build script does all of the RattlesnakeOS-stack specific stuff *after* the user stuff, so that order could be preserved in this type of setup. But unlike the current setup, if a user specifically wants their stuff applied/run *after* the RattlesnakeOS-specific stuff, they would have the power to do so.
- There are a few instances where build_template.go does some file shuffling to ensure files make it into the right place (i.e. moving Chromium apks into the build). I think these could be converted into "custom scripts."
- Things such as this requiring AWS credentials would require some kind of "bootstrapping" or "env setup" to take place. Assuming this proposed feature of aosp-build running custom scripts keeps every script running under the same environment, you could have a script that initializes the environment for whatever you need to do in AWS, and another custom script to handle keys, another to upload to S3, and etc. If something needs to run in its own environment, such as for kernel builds, each individual script can be written so that actually happens (and since aosp-build requires Docker, perhaps that can be leveraged in some way for environment-sensitive tasks).
- Doing things this way, to me, makes things less complex, in the sense that you don't have to maintain a big complex build script alongside the script in aosp-build's master. The "essence" of what makes RattlesnakeOS-stack produce RattlesnakeOS ROMs gets translated into these scripts, and all you have to maintain is that.
- This also lets end users get a little more hands on with the build process without actually having to touch the main build scripts. What would have previously required "hacking" build_template.go to modify the RattlesnakeOS-stack build process, can now be achieved by slotting in a custom script in-between certain custom scripts that do special RattlesnakeOS-type things.
[... cut ...]
One thing that confuses me a bit at the moment is figuring out how to update a user's local config.yml, with all of their customizations, every month, without requiring them to write a new one each month. Then again, not a lot would need to change every month other than the build tag, so that could probably be silently or interactively updated for the user when they run `rattlesnakeos-stack deploy`

---

It looks like I got carried away and dumped my brain for you again. You have quite a lot to read here - sorry about that 😬

ubergeek77, 2019-12-29:
So, I've been trying to get aosp-build up and running so I can test porting some RattlesnakeOS stuff to it. Unfortunately, "trying" is the operative word here.

I don't know what it is about how aosp-build pulls the Android source, but I literally cannot get it to successfully finish a pull. Every time I try, it will download a bunch of objects, but at some point it will be interrupted with a "remote end hung up" error and terminate the pull. After fighting with it hands-on to get it to pull, I resigned to running a script by trapping `make fetch` in a bash `until` line, so it would continue to retry after a fail (and this is *on top of* the retry provisions inside of the fetch script). Even after letting this run for **three days**, trying to fetch over and over, the pull still wasn't finished.

At one point, I was able to get `make fetch` complete with an exit code of 0 (success), since it broke out of the until loop. However, the build failed, since critical build directories were empty. For example, frameworks/base was empty apart from the hidden .directories.

Now, I don't have particularly bad internet, either. This is just something about how aosp-build handles the pull.

When I take your build script from RattlesnakeOS, take out the AWS stuff, and get it to pull locally and start a local build, it does so with absolutely no problems whatsoever. I was able to get the Android source pulled in about 20 minutes or so, and the build started right after. I've had no problems with your script.

But with aosp-build, I seriously cannot get it to behave no matter what I try. When run locally, even after 3 days, I can't get it to do something simple like pull down Android's source code. It seems to work fine when run on an AWS VM with datacenter internet, but there's something about "reliable-but-sometimes-spotty" residential internet that aosp-build just isn't compatible with (but your build script is, apparently).

I can 100% reproduce this on my end, as I've been toying with this over multiple days since I wrote my last reply here. I never have any issues with running your build script locally (even when running from scratch with a clean directory), but something always goes wrong with aosp-build, on the same machine, on the same OS.

I'm not sure if this belongs on an issue in aosp-build or not, but I think it's relevant here, since local RattlesnakeOS builds with aosp-build are on the table in the future.

Do you have any ideas as to why aosp-build is giving me so much trouble? Maybe aosp-build's pull method could be made more similar to how your script handles it.

I'd be happy to run any tests on my end, if you like. This is really perplexing to me, since aosp-build seems very well put together, so I really can't understand why I'm having so much trouble with it.

dan-v, 2019-12-30:
@ubergeek77 - sorry for the radio silence; I've been reading your comments but haven't had a chance to reply. I don't have time to post in great detail at the moment, but here is my brief response:

As for your first post, I think you have a great point on keeping the same methodology of providing only patch/script/prebuilt repositories and not having to maintain a whole repo. And reading through your bullet points, I'm generally onboard with what you are proposing the end to end aosp-build integration would look like.

As for your second post, that is really strange that you are having issues from your home setup with aosp-build fetch vs modified RattlesnakeOS script as it's essentially running the same thing (`repo sync`). I'm curious if maybe it's a docker version or configuration issue that is causing issues on your home setup? I've definitely seen docker performance issues vary depending on OS, storage plugins, platform, etc. Maybe a worthy test would be executing your modified RattlesnakeOS build script in a docker container on this same host and see if it has similar problems with sync. I haven't attempted a sync with either approach at home though, so I don't have any additional datapoints to look at unfortunately. I have only verified aosp-build sync worked fine within AWS.

Rudd-O, 2020-01-05:
RattlesnakeOS-stack works fine here at home when running modified under Jenkins.  My net connection isn't quite spotty, but it isn't datacenter quality, and I sure cannot get `repo sync` to fail or end up with incomplete syncs.

ubergeek77, 2020-01-05:
> 
> 
> RattlesnakeOS-stack works fine here at home when running modified under Jenkins. My net connection isn't quite spotty, but it isn't datacenter quality, and I sure cannot get `repo sync` to fail or end up with incomplete syncs.

Exactly - that was my experience too. RattlesnakeOS-stack's build script has been perfectly stable for me, even after being amputated to remove anything too AWS specific, and to allow it to run locally.

But I can't say the same about [aosp-build](https://github.com/hashbang/aosp-build), which Dan might migrate to at some point. My issue was that I can't get aosp-build to behave, because it seems to not be very stable about `repo sync`.

One thing I did notice when trying aosp-build is that it pulls different tags for everything for some reason. Despite `android-10.0.0_r15` being the latest tag for `crosshatch`, it seems to pull manifests for `android-10.0.0_r17`. This is after running `make DEVICE=crosshatch config manifest` as @dan-v suggested.

Maybe that command just generates a screwy manifest? I think RattlesnakeOS-stack just pulls one directly from Google, so maybe that's why it doesn't `repo sync` properly.

But I'd love to hear if you are able to get aosp-build to work by doing exactly what I did. Maybe it could have been a local Docker network anomaly like Dan mentioned - I haven't really tested that yet.

dan-v, 2020-01-22:
@ubergeek77 - sorry I haven't had a chance to play around with aosp-build recently, but will let you know when I do. Thanks for taking a first look at it and for documenting some issues to investigate.

ubergeek77, 2020-01-22:
@dan-v It's my pleasure!

I did end up getting it going, but I gave up on trying a normal repo sync and just made a local AOSP mirror on my NAS instead. This is a better long-term solution for me in general, and it makes sense to have, with or without RattlesnakeOS, since I find myself creating builds so frequently. A full, default AOSP mirror is a surprisingly tame 380GB, and I could probably knock that down by a lot if I were to remove stuff from devices I don't need. In addition, it turns out a few of the earlier issues I had were just bugs in aosp-build, which I've since reported, and are now pending in a PR from a contributor over there.

As far as integrating it with RattlesnakeOS-stack - I certainly see some benefit, but one thing I don't understand is why creating a manifest on aosp-build takes so long. It takes at least 5 minutes to just generate a manifest, which is something I don't think RattlesnakeOS-stack does at all. It seems like RattlesnakeOS-stack can eloquently pull and build against a given tag (i.e. android-10.0.0_r21), without needing to manually and recursively find and specify *every* commit hash for every project associated with that tag (there's also a bug in how it handles this, which I have also reported).

Maybe it makes sense from a technical standpoint (though admittedly I don't see how), but as a user, I don't understand why aosp-build has to be slower by design when generating a manifest for a repo sync, when compared to the current RattlesnakeOS-stack. But cutting that time down is definitely high on my list of "wants," should RattlesnakeOS ever fully switch to it.

Rudd-O, 2020-02-25:
What method did you use to create the mirror, @ubergeek77 ?  Thanks in
advance.

ubergeek77, 2020-02-28:
@Rudd-O Assuming you already have `repo` installed, you can run this to initialize a new mirror:

`repo init -u https://android.googlesource.com/mirror/manifest`

And then simply `repo sync` while in that directory. You may want to run this about once a month. Sometimes, a few sources don't *always* get picked up (not often, but it can happen), so I like to set it to run in sequence 3 times in a row on that same day, that way everything gets picked up. For your initial sync, you'll know you're done mirroring when nothing else shows as having been synced.

Note that this will sync *literally everything that has ever been committed in the history of AOSP.* That's manageable for me, personally, as it's only around 422GB as of right now and I have the storage space for it. To cut down on size, I'm sure you could muck about with the auto-generated manifest files and remove stuff that is obviously never going to be used (such as vendor blobs for really old devices), but I guess I'm just too lazy to do that. It's worth noting that the size has increased by over 35GB in the last two weeks, so some size reduction measures may be a good idea in the near future.

For anyone else coming across this issue and wanting to help RattlesnakeOS with this migration, I'd like to say that starting your own AOSP mirror is a great way to trade the up front time spent syncing the mirror for time savings when you want to do clean builds later. It's ***much*** easier than manually syncing from upstream AOSP every time you want to test something.

dan-v, 2021-03-01:
Closing this, may reconsider this if aosp-build is ever revived.

closed 2021-03-01 by dan-v