dan-v/rattlesnakeos-stack #147: Custom Script for boot animation not working and option to change default screenlock wallpaper

BinoopV-Contus, 2020-01-09:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options

### Description
Custom script executed for the boot animation is throwing error and and build gets failed. 

### Full Config Used for Setup
attestation-instance-type = "t3.nano"
attestation-max-price = ".005"
attestation-server = false
chromium-version = ""
device = "bonito"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
region = "ap-south-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "rattlesnake"

[[custom-scripts]]
  repo = "https://github.com/RattlesnakeOS/example_patch_shellscript"
  scripts = [ "00002-custom-boot-animation.sh" ]

### Email Notification Output
RattlesnakeOS Build FAILED
  Device: bonito
  Stack Version: 10.0.1
  Stack Region: ap-south-1
  Release Channel: bonito-stable
  Instance Type: c5.4xlarge
  Instance Region: us-east-2
  Instance IP: 18.222.153.182
  Build Date: 2020.01.09.12
  Elapsed Time: 0hrs 17min 38sec
  AOSP Build: QQ1A.200105.002
  AOSP Branch: android-10.0.0_r21
  Chromium Version: 81.0.4000.0
  F-Droid Version: 1.7.1
  F-Droid Priv Extension Version: 0.2.9
  Build Reason: No build is required, but FORCE_BUILD=true
.......... 10%  392M 1s
  1300K .......... .......... .......... .......... .......... 11%  379M 1s
  1350K .......... .......... .......... .......... .......... 11%  341M 0s
  1400K .......... .......... .......... .......... .......... 12%  383M 0s
  1450K .......... .......... .......... .......... .......... 12%  392M 0s
  1500K .......... .......... .......... .......... .......... 13%  358M 0s
  1550K .......... .......... .......... .......... .......... 13%  318M 0s
  1600K .......... .......... .......... .......... .......... 13%  336M 0s
  1650K .......... .......... .......... .......... .......... 14% 6.13M 0s
  1700K .......... .......... .......... .......... .......... 14%  204M 0s
  1750K .......... .......... .......... .......... .......... 15%  235M 0s
  1800K .......... .......... .......... .......... .......... 15%  383M 0s
  1850K .......... .......... .......... .......... .......... 15%  391M 0s
  1900K .......... .......... .......... .......... .......... 16%  273M 0s
  1950K .......... .......... .......... .......... .......... 16%  322M 0s
  2000K .......... .......... .......... .......... .......... 17%  379M 0s
  2050K .......... .......... .......... .......... .......... 17%  293M 0s
  2100K .......... .......... .......... .......... .......... 18%  385M 0s
  2150K .......... .......... .......... .......... .......... 18%  341M 0s
  2200K .......... .......... .......... .......... .......... 18% 64.5M 0s
  2250K .......... .......... .......... .......... .......... 19%  275M 0s
  2300K .......... .......... .......... .......... .......... 19%  203M 0s
  2350K .......... .......... .......... .......... .......... 20%  243M 0s
  2400K .......... .......... .......... .......... .......... 20%  396M 0s
[... 169 lines cut ...]
 10900K .......... .......... .......... .......... .......... 91%  288M 0s
 10950K .......... .......... .......... .......... .......... 92%  186M 0s
 11000K .......... .......... .......... .......... .......... 92%  232M 0s
 11050K .......... .......... .......... .......... .......... 93%  184M 0s
 11100K .......... .......... .......... .......... .......... 93%  249M 0s
 11150K .......... .......... .......... .......... .......... 94%  176M 0s
 11200K .......... .......... .......... .......... .......... 94%  278M 0s
 11250K .......... .......... .......... .......... .......... 94%  382M 0s
 11300K .......... .......... .......... .......... .......... 95%  404M 0s
 11350K .......... .......... .......... .......... .......... 95%  351M 0s
 11400K .......... .......... .......... .......... .......... 96%  385M 0s
 11450K .......... .......... .......... .......... .......... 96%  384M 0s
 11500K .......... .......... .......... .......... .......... 97%  395M 0s
 11550K .......... .......... .......... .......... .......... 97%  319M 0s
 11600K .......... .......... .......... .......... .......... 97%  381M 0s
 11650K .......... .......... .......... .......... .......... 98%  394M 0s
 11700K .......... .......... .......... .......... .......... 98%  376M 0s
 11750K .......... .......... .......... .......... .......... 99%  345M 0s
 11800K .......... .......... .......... .......... .......... 99%  379M 0s
 11850K .......... .......... .......... .......... .......... 99%  388M 0s
 11900K ...                                                   100% 5867G=0.1s

2020-01-09 12:57:25 (80.4 MB/s) - ‘/home/ubuntu/rattlesnake-os/branding/bootanimation.zip’ saved 

sed: can't read /home/ubuntu/rattlesnake-os/build/make/target/product/core.mk: No such file or directory

dan-v, 2020-01-09:
Yeah, this was working on 9.0 but is not in 10.0. The `core.mk` file doesn't exist anymore (https://android.googlesource.com/platform/build/+/refs/tags/android-10.0.0_r21/target/product/). I just [committed a change](https://github.com/RattlesnakeOS/example_patch_shellscript/commit/043e453d73472cbb7cceee51d02566445ed4adf0) to use the same file as build script (`handheld_system.mk`), but don't have time to test this. Feel free to re-open if still not working.

closed 2020-01-09 by dan-v

BinoopV-Contus, 2020-01-11:
@dan-v : It works well. If possible could you guide me how to change the default wallpaper of rattlesnake OS. I referred several but unable to find a solution on how to implement it

dan-v, 2020-01-12:
@BinoopV-Contus - unfortunately I don't have time to help with that. The high level approach could be done with a script and would download your wallpaper and overwrite the existing one (default_wallpaper.png) that lives in https://android.googlesource.com/platform/frameworks/base/.