dan-v/rattlesnakeos-stack #148: get_latest_versions() doesn't return latest F-Droid Privileged Extension version

svcraig, 2020-01-22:
### Prerequisites

* [ X] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [ X] I am able to reproduce this issue without any advanced customization options

### Description

FDROID_PRIV_EXT_VERSION is continually set to 0.2.9 in the get_latest_versions function of the build script despite 0.2.11 being the latest non-alpha, non-OTA version.

### Steps to Reproduce

1. In a Linux shell, run the relevant part of the function, substituting the correct URL:

`curl --fail -s "https://gitlab.com/api/v4/projects/1481578/repository/tags" | jq -r 'sort_by(.name) | reverse | [.[] | select(.name | test("^[0-9]+\\.[0-9]+")) | select(.name | contains("alpha") | not) | select(.name | contains("ota") | not)][0] | .name'`

**Expected behavior:** Observe the output 0.2.11

**Actual behavior:**  Observe the output 0.2.9

The regex isn't treating .10 and .11 as 10 and 11

I'm not familiar enough with regex to solve this properly, but I can offer a hacky solution: sort by committed_date instead of name, and don't reverse.

The command

`curl --fail -s "https://gitlab.com/api/v4/projects/1481578/repository/tags" | jq -r 'sort_by(.committed_date) | [.[] | select(.name | test("^[0-9]+\\.[0-9]+")) | select(.name | contains("alpha") | not) | select(.name | contains("ota") | not)][0] | .name'`

Yields the correct output 0.2.11. As long as the author doesn't randomly decide to commit an older numbered version, it just might be okay...

dan-v, 2020-01-23:
Thanks for the report @svcraig! I was aware of this issue, but didn't put in a separate bug for it. I was hoping to fix it at the same time as #142, but even on the latest f-droid versions I'm still seeing issues with privileged extension installs.

btw, this was the version of the fix I came up with (haven't pushed commit yet); it would be good to have another person look at it for validation though:
`
$(curl --fail -s "$FDROID_PRIV_EXT_URL_LATEST" | jq -r '[.[] | select(.name | test("^[0-9]+\\.[0-9]+")) | select(.name | contains("alpha") | not) | select(.name | contains("ota") | not)][] | .name' | sort --version-sort -r | head -1)
`

svcraig, 2020-01-23:
@dan-v that fixes it! I too am having issues with the priv extension,  which is what got me digging down this rabbit hole. The trouble is I'm brand new to F-Droid so I'm not familiar with the expected behaviours. The extension seems to be correctly signed but I still see an error in logcat about being unable to set the installer package due to the seller's signature not matching

dan-v, 2020-01-24:
@svcraig - there's an ongoing thread here for the issue: https://gitlab.com/fdroid/privileged-extension/issues/47#note_276186017. @chirayudesai has committed a number of fixes around this related to permission issues already and he has tracked down one more issue the other day. so hopefully once that gets merged and lands in a release, the privileged extension will be working as intended again.

svcraig, 2020-01-27:
@dan-v thanks for linking me with that issue. I _think_ the latest commit fixes the issue for me, but I had to revert some other commits to make sure, which leads to another question:

How do you want to handle bug reports for community patches, as they're beyond the scope of your code? [Patch 1](https://github.com/RattlesnakeOS/community_patches/blob/10.0/00001-global-internet-permission-toggle.patch) seems like it's for Pie only, as the permission handling was massively reworked for Q.

As a side note, GrapheneOS has a few commits for a global internet permission on Q. I was hoping to test them out and open a merge request to include them....but I suspect the new permission was causing another issue with the privileged extension. I'll track that down. If so, it might mean there was a gap between people using the permission and people using F-Droid, and the permission AOSP patch will have a dependency on a (new) related F-Droid client patch.

dan-v, 2020-02-03:
Should be resolved in [v10.0.2](https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v10.0.2) Note it is pulling in alpha versions as well for the client for the time being.

@svcraig as for your question on patches, bugs for these patches should go in the community repo but going from 9.0 to 10.0 the patches were mostly not migrated so many were not working. Recently, there has been some great contributions from @hackoder to help update the repo for 10.0 and mark what ones are confirmed as working as well.

closed 2020-02-03 by dan-v