dan-v/rattlesnakeos-stack #150: How to remove Fdroid from Rattlesnake OS ?

BinoopV-Contus, 2020-01-29:
I am trying to remove the Froid from the Rattlesnake OS build, How is it possible to achieve it when building the OS

dan-v, 2020-02-03:
Currently there is no easy or clean way to do this, but with a customization script you could remove the fdroid client and privileged extension from being included in the build by doing something like sed on the file where these packages are included (https://github.com/dan-v/rattlesnakeos-stack/blob/10.0/templates/build_template.go#L890). Fdroid client and privileged extension would still be built and would still be part of the version checks which determine if a new build should be executed. Anyways, I'll turn this issue into a feature request as I think F-droid could be optional.

BinoopV-Contus, 2020-02-03:
Thank you @dan-v , right now I have achieved by rebuilding the stack by removing the Fdroid build process.

dan-v, 2021-03-01:
This will now be possible by using the new core config repo option and just not including F-Droid.

closed 2021-03-01 by dan-v