dan-v/rattlesnakeos-stack #151: Initial build failed due to: chrome/common/buildflags.h' file not found

laamalif, 2020-01-30:
### Prerequisites

* [ ] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [ ] I am able to reproduce this issue without any advanced customization options

### Description

Initial build failed due to: chrome/common/buildflags.h' file not found

Tested chromium versions: 81.0.4040.6, 80.0.3971.4, 80.0.3987.78

### Steps to Reproduce
### Platform
Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-instance-type = "t3.nano"
attestation-max-price = ".005"
attestation-server = false
chromium-version = "80.0.3971.4"
device = "crosshatch"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.18xlarge"
max-price = "1.00"
name = "rattlesnakeos-crosshatch-xzyz"
region = "us-east-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.68"
ssh-key = "rattlesnakeos"

[[custom-patches]]
patches = ["00002-microg-sigspoof.patch"]
repo = "https://github.com/RattlesnakeOS/microg"

[[custom-prebuilts]]
modules = ["GmsCore","GsfProxy","FakeStore","com.google.android.maps.jar"]
repo = "https://github.com/RattlesnakeOS/microg"

```

### Email Notification Output
Should look something like this. You can mask stack name from here and you may want to verify contents of log output before pasting.
```
  Device: crosshatch
  Stack Name: rattlesnakeos-crosshatch-mtumhnc4hq
  Stack Version: 10.0.1
  Stack Region: us-east-1
  Release Channel: crosshatch-stable
  Instance Type: c5.18xlarge
  Instance Region: us-east-2
  Instance IP: 18.216.174.85
  Build Date: 2020.01.30.14
  Elapsed Time: 1hrs 20min 54sec
  AOSP Build: QQ1A.200105.002
  AOSP Branch: android-10.0.0_r21
  Chromium Version: 80.0.3987.78
  F-Droid Version: 1.7.1
  F-Droid Priv Extension Version: 0.2.9
  Build Reason: Initial build
...
[320/8823] CXX obj/chrome/browser/updates/announcement_notification/announcement_notification/announcement_notification_service.o
FAILED: obj/chrome/browser/updates/announcement_notification/announcement_notification/announcement_notification_service.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/chrome/browser/updates/announcement_notification/announcement_notification/announcement_notification_service.o.d -DOFFICIAL_BUILD -DNO_UNWIND_TABLES -D_GNU_SOURCE -DANDROID -DHAVE_SYS_UIO_H -DANDROID_NDK_VERSION_ROLL=r20_1 -DCR_CLANG_REVISION=\"n332890-c2443155-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBP_EXTERN=extern -DVK_NO_PROTOTYPES -DUSE_EGL -DSK_HAS_PNG_LIBRARY -DSK_HAS_WEBP_LIBRARY -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_GL -DSK_HAS_JPEG_LIBRARY -DSK_USE_LIBGIFCODEC -DSK_VULKAN_HEADER=\"../../skia/config/SkVulkanConfig.h\" -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_BUILD_FOR_ANDROID -DUSE_CHROMIUM_SKIA -DVK_NO_PROTOTYPES -I../.. -Igen -I../../third_party/libyuv/include -I../../third_party/jsoncpp/source/include -I../../third_party/jsoncpp/generated -I../../third_party/libwebp/src -I../../third_party/vulkan/include -I../../third_party/khronos -I../../gpu -I../../third_party/perfetto/include -Igen/third_party/perfetto/build_config -Igen/third_party/dawn/src/include -I../../third_party/dawn/src/include -I../../third_party/boringssl/src/include -I../../third_party/skia -I../../third_party/libgifcodec -I../../third_party/vulkan/include -I../../third_party/skia/third_party/vulkanmemoryallocator -I../../third_party/vulkan/include -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fprofile-sample-accurate -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fcomplete-member-pointers -ffunction-sections -fno-short-enums --target=aarch64-linux-android21 -mno-outline -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Werror -Wextra -Wimplicit-fallthrough -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-bitwise-conditional-parentheses -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Oz -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g1 -fdebug-info-for-profiling -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -std=c++14 -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../third_party/android_ndk/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fvisibility-inlines-hidden -Wno-deprecated-declarations -c ../../chrome/browser/updates/announcement_notification/announcement_notification_service.cc -o obj/chrome/browser/updates/announcement_notification/announcement_notification/announcement_notification_service.o
In file included from ../../chrome/browser/updates/announcement_notification/announcement_notification_service.cc:12:
../../chrome/browser/browser_process.h:22:10: fatal error: 'chrome/common/buildflags.h' file not found
[... 196 lines cut ...]
Completed 6.1 MiB/10.8 MiB (21.4 MiB/s) with 1 file(s) remaining
Completed 6.3 MiB/10.8 MiB (22.2 MiB/s) with 1 file(s) remaining
Completed 6.6 MiB/10.8 MiB (23.0 MiB/s) with 1 file(s) remaining
Completed 6.8 MiB/10.8 MiB (23.8 MiB/s) with 1 file(s) remaining
Completed 7.1 MiB/10.8 MiB (24.1 MiB/s) with 1 file(s) remaining
Completed 7.3 MiB/10.8 MiB (24.8 MiB/s) with 1 file(s) remaining
Completed 7.6 MiB/10.8 MiB (25.6 MiB/s) with 1 file(s) remaining
Completed 7.8 MiB/10.8 MiB (26.3 MiB/s) with 1 file(s) remaining
Completed 8.1 MiB/10.8 MiB (26.8 MiB/s) with 1 file(s) remaining
Completed 8.3 MiB/10.8 MiB (27.2 MiB/s) with 1 file(s) remaining
Completed 8.6 MiB/10.8 MiB (27.9 MiB/s) with 1 file(s) remaining
Completed 8.8 MiB/10.8 MiB (28.6 MiB/s) with 1 file(s) remaining
Completed 9.1 MiB/10.8 MiB (29.3 MiB/s) with 1 file(s) remaining
Completed 9.3 MiB/10.8 MiB (30.0 MiB/s) with 1 file(s) remaining
Completed 9.6 MiB/10.8 MiB (30.8 MiB/s) with 1 file(s) remaining
Completed 9.8 MiB/10.8 MiB (30.0 MiB/s) with 1 file(s) remaining
Completed 10.1 MiB/10.8 MiB (30.7 MiB/s) with 1 file(s) remaining
Completed 10.3 MiB/10.8 MiB (31.3 MiB/s) with 1 file(s) remaining
Completed 10.6 MiB/10.8 MiB (32.0 MiB/s) with 1 file(s) remaining
Completed 10.8 MiB/10.8 MiB (32.7 MiB/s) with 1 file(s) remaining
upload: ../../../../var/log/cloud-init-output.log to s3://rattlesnakeos-crosshatch-mtumhnc4hq-logs/crosshatch/1580399668
==================================
2020-01-30 15:54:28: Running aws_notify
==================================
```

dan-v, 2020-01-31:
Thanks for the report @laamalif. I haven't attempted a Chromium build in a few weeks, but will kick something off and try to debug. I was previously able to build and am running '81.0.4027.2' if you want to give that a shot in the mean time.

dan-v, 2020-01-31:
I was unfortunately not able to reproduce this issue, my latest build with Chromium `81.0.4040.6` went through fine. 

```
RattlesnakeOS Build SUCCESS
 Device: crosshatch
 Stack Version: 10.0.1
 Stack Region: us-west-2
 Release Channel: crosshatch-stable
 Instance Type: c5.18xlarge
 Instance Region: us-east-2
 Instance IP: 18.224.183.124
 Build Date: 2020.01.31.04
 Elapsed Time: 2hrs 32min 53sec
 AOSP Build: QQ1A.200105.002
 AOSP Branch: android-10.0.0_r21
 Chromium Version: 81.0.4040.6
 F-Droid Version: 1.8-alpha1
 F-Droid Priv Extension Version: 0.2.11
 Build Reason:  'Chromium version 81.0.4027.2 != 81.0.4040.6'
```
* note: i have one uncommited change to fix pulling in latest f-droid versions, but nothing related to chromium.

pgera, 2020-01-31:
I am seeing this with 81.0.4027.2 as well.

dan-v, 2020-01-31:
Hmm.. I've had no issues building `81.0.4027.2` or latest `81.0.4040.6`. Normally Chromium build issues like these are easy to reproduce. The only difference between our builds appears to be microg patches, but not sure how that would affect the Chromium build process.

corrmaan, 2020-02-02:
I experienced the same error last night with version `81.0.4040.6`, vanilla RattlesnakeOS with a brand new config and deployment. No special changes. Do you think it would work if I changed `CHROME_CHANNEL` to stable?

[2020-02-02-024734-build.log](https://github.com/dan-v/rattlesnakeos-stack/files/4143872/2020-02-02-024734-build.log)

pgera, 2020-02-02:
Stable is still on 79, whereas Q needs 80 or newer. Beta is on 80 though. So instead of dev, it may be worth trying beta (and stable in a few days when 80 hits stable).

corrmaan, 2020-02-02:
Maybe it has to do with too much build parallelization? I'll try a new build with `--jobs` in Chromium's build hard set to something like 4 or 8 instead of `c5.4xlarge`'s 16 from `nproc`.

dan-v, 2020-02-02:
@corrmaan - I was also thinking it could be related to build parallelization, but the original report was on a `c5.18xlarge` which has many more cores. So far I've had 1 successful build on `81.0.4027.2` and 2 successful builds on `81.0.4040.6` all on a `c5.18xlarge` instances. I'll try to kick off a build on a smaller instance type to see if I can reproduce.

corrmaan, 2020-02-02:
Sounds good. My comment above was just a guess on how the Chromium build operates - after looking in to it it seems to me like any `--jobs` settings are handled by `autoninja` and wrapped up in some python scripts.

dan-v, 2020-02-02:
Could also be related to build caching as well since the webview is being built first and then the browser is being built without cleaning first (this was intended to reduce build time). It would be interesting to see if this could reproduce if:
* the build directory was wiped between the build of webview and browser (this would make the build times very bad)
* the order of build was swapped to browser first and then webview 

I'm still just trying to get a solid repro at this point; I kicked off a c5.4xlarge build and hoping I bump into the issue easier with that.

dan-v, 2020-02-02:
Nice, I was able to reproduce this issue with a c5.4xlarge. So the question becomes what to do to fix it. I could:
* disable building/adding chromium browser altogether and only include the webview. i would probably want to keep aosp browser in that case so people on new builds had an initial browser.
* clean between webview and browser build, but this would double build time
* try to get trichrome to work so there aren't two separate builds for webview and browser

In the mean time, I'm going to play around with swapping the build order to see what happens in that case.

dan-v, 2020-02-03:
Looks like swapping the build order to first build the browser and then webview was successful on c5.4xlarge. I might do that at least temporarily until I figure out how to properly fix this situation. I'll try to get a new version out soon with that change.

corrmaan, 2020-02-03:
Nice, thanks @dan-v.
I was going to say that removing Chromium entirely seems like a step backwards. At least maybe instead just inclide an option to disable it. 
Cleaning between builds seems reasonable enough. Trichrome sounds like a good direction to head in too though. Whatever is easiest!

dan-v, 2020-02-03:
Pushed a new [release](https://github.com/dan-v/rattlesnakeos-stack/releases) with the ordering workaround - hopefully it holds up for the time being. If someone gets a chance to test it out and could report back that would be a great help.

pgera, 2020-02-03:
I can confirm that the build passed. Slight OT, but wanted to check if anyone has this chromium issue. On Q, I have never been able to download files in chromium. For eg., downloading a test pdf file (https://www.w3.org/WAI/ER/tests/xhtml/testfiles/resources/pdf/dummy.pdf) throws an "SD card not found" error. I wonder if this is also microg related somehow as that is the only deviation in my build.

corrmaan, 2020-02-03:
Thanks @dan-v, that worked for me too.
Sorry @pgera but I haven't ran in to that issue before, but maybe I've never tried downloading files on Chromium either.

laamalif, 2020-02-03:
Location selection doesnt work on chromium while downloading files. I've also noticed that chromium keep crashing on video playback other than youtube site. maybe codec issue?

dan-v, 2020-02-04:
@pgera - I can confirm that there appears to be an issue with that. It seems like it downloads it but it says 'failed to open' and it doesn't appear to be on disk. Not entirely sure what this issue is.

@laamalif - The crash on video playback issue is likely the same as seen here: https://github.com/GrapheneOS/Vanadium/pull/52. It sounds like there is a patch that works around the primary issue, but right now I don't support patches of Chromium. I'm gonna wait and see how the issue plays out for now.

laamalif, 2020-02-05:
Do you have plan to support patches for Chromium in future? I think having minimal or webview customization will be good option, in the end you still need another browser for day to day task.

Chromium patch support will be nice feature to have in my opinion or at least customization of webview only build.

dan-v, 2020-02-06:
@laamalif - i have an enhancement open for it https://github.com/dan-v/rattlesnakeos-stack/issues/104, but I haven't had much time to implement new features lately and have been in maintenance mode for the most part (hopefully that changes in the future as there's a lot i'd like to add/change with this project). Anyways, for now I'm happy to take PRs for enhancements like this.

dan-v, 2020-02-06:
Closing this and opened up an enhancement to move to Trichrome (https://github.com/dan-v/rattlesnakeos-stack/issues/152)

closed 2020-02-06 by dan-v

pgera, 2020-02-10:
The download failure is being tracked upstream at https://bugs.chromium.org/p/chromium/issues/detail?id=1045642

and downstream at:
https://github.com/GrapheneOS/Vanadium/issues/56
https://github.com/bromite/bromite/issues/485