dan-v/rattlesnakeos-stack #153: Simplify release tracking with a text file

pgera, 2020-03-10:
Currently, the latest release information is fetched from google's release pages and a couple of regex rules are used to derive the latest release information. This has a couple of issues:

- Regex parsing can be fragile which can lead to unexpected issues such as stale or incorrect version being built.
- In some cases, the regex logic may return a build whose sources have not dropped. I haven't confirmed if this is still an issue, but I have seen this in the past.

The proposal is to push a txt/toml/json file to the repo every month with the most recent build tag. This file can be polled instead of the regex logic to fetch the build info. This also fixes the 2nd point above since it will always build something successfully, even if it is the previous month's build in cases where the source for the most recent build is not available.

dan-v, 2020-05-10:
@pgera / @scott-8 - my current plan is to also put fdroid and chromium versions in this repo in addition to AOSP build/tag. here's the current layout: https://github.com/RattlesnakeOS/latest. let me know if you have any feedback on that.

pgera, 2020-05-10:
Looks good. Since chromium updates about twice a month, it may become cumbersome to test/update the tags that often. Is the plan to update everything once a month at release time ?

dan-v, 2020-05-10:
@pgera - yeah, I was planning on updating once a month at release time, and maybe if there was some important update it could be updated out of band.

commit 48b2022f (2020-05-10): change over to using https://github.com/RattlesnakeOS/latest to lookup latest aosp build/tag, chromium, fdroid versions. closes #153
closed 2020-05-10 by dan-v