dan-v/rattlesnakeos-stack #158: Updater shows unwanted notifications for transient failures

nickrbogdanov, 2020-04-30:
https://github.com/RattlesnakeOS/platform_packages_apps_Updater/commit/45f10ce8c5b851f15f3eb6c9fe031c6d82f34dbe added a notification for failed updates. One of the side effects of this change is that in the middle of the night, when my Comcast internet is down for maintenance, my phone wakes me up to tell me:

```
Error updating OS
Please check your settings.
Error details: Unable to resolve host "rattlesnakeos-XXX-release.s3.amazonaws.com": No address associated with hostname
```

I think the desired behavior is more along the lines of:

 * If a background update fails due to connectivity issues, it should fail silently and just retry in a little bit.
 * Unless (maybe) it's tried for several days and it keeps failing. Then the user probably wants to know about it.
 * If a _forced_ update fails to download for any reason, it's probably good to tell the user why, because they're expecting the update. Maybe this should be in a notification or maybe it should just be in the Settings pane.
 * If an update downloads successfully but fails to install for any reason, that should definitely be reported to the user. Since this is something that should never happen.

dan-v, 2020-04-30:
@nickrbogdanov - I've also bumped into this when I had spotty cell coverage. I think it would make sense to just not notify/check if there is no internet connectivity. Thoughts?

scott-8, 2020-05-05:
I think that @dan-v's idea is probably the best and the easiest to implement. One thing to note is that the entire update process is wrapped in a try...catch block, and singling out any specific errors would make it more complicated and move the code farther from upstream.
 
Another easy thing to do would be to remove vibrations from the notification (although I would argue that the notifications are fairly urgent in cases like #144, since the updater retries every 4 minutes).

scott-8, 2020-05-06:
I added [this](https://github.com/scott-8/platform_packages_apps_Updater/commit/46cce29de62ec6470c8d3a973972594f0a073706), assuming `fetchData` is the method that fails.

dan-v, 2020-05-06:
thanks @scott-8 - this seems like a relatively simple workaround that won't require deviating too far from upstream. i'd like to do a bit of validation before pulling this in as breaking the updater is never fun.

closed 2020-06-19 by dan-v