dan-v/rattlesnakeos-stack #159: Script fails to get build date

scott-8, 2020-05-10:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options

### Description

Can't get latest factory build date.

### Steps to Reproduce

Build AOSP.

**Expected behavior:** It to build.

**Actual behavior:** It doesn't build.

### Platform
Linux.

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-instance-type = "t3.nano"
attestation-max-price = ".005"
attestation-server = false
chromium-version = ""
device = "sargo"
email = ""
encrypted-keys = false
hosts-file = "https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts"
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = ""
region = "us-east-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "rattlesnakeos"
```

### Carrier
Doesn't matter.

### Email Notification Output
Should look something like this. You can mask stack name from here and you may want to verify contents of log output before pasting.
```
RattlesnakeOS Build FAILED
  Device: sargo
  Stack Name: 
  Stack Version: 10.0.7
  Stack Region: us-east-1
  Release Channel: sargo-stable
  Instance Type: c5.4xlarge
  Instance Region: us-east-2
  Instance IP: 
  Build Date: 2020.05.10.03
  Elapsed Time: 0hrs 0min 4sec
  AOSP Build: 
  AOSP Vendor Build:
  AOSP Branch:
  Chromium Version: 81.0.4044.138
  F-Droid Version: 1.8-alpha2
  F-Droid Priv Extension Version: 0.2.11
  Build Reason:
.
.
.
.
.
.
.
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 59832 files and directories currently installed.)
Preparing to unpack .../libonig4_6.7.0-1_amd64.deb ...
Unpacking libonig4:amd64 (6.7.0-1) ...
Selecting previously unselected package libjq1:amd64.
Preparing to unpack .../libjq1_1.5+dfsg-2_amd64.deb ...
Unpacking libjq1:amd64 (1.5+dfsg-2) ...
Selecting previously unselected package jq.
Preparing to unpack .../jq_1.5+dfsg-2_amd64.deb ...
Unpacking jq (1.5+dfsg-2) ...
Setting up libonig4:amd64 (6.7.0-1) ...
Setting up libjq1:amd64 (1.5+dfsg-2) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Setting up jq (1.5+dfsg-2) ...
Running the latest rattlesnakeos-stack version 10.0.7
Searching for latest factory build date for device=sargo android_version=10.0
==================================
2020-05-10 03:55:26: Running aws_notify_simple
==================================
.
.
.
.

```

### Full log
N/A

scott-8, 2020-05-10:
It looks like Google the "Acknowledge" box to https://developers.google.com/android/images, so parsing the page fails.

scott-8, 2020-05-10:
Ok I added `--cookie "devsite_wall_acks=nexus-image-tos"` to lines 193 and 201 in the script and the build started successfully (I'll update this tomorrow once it's done building).

dan-v, 2020-05-10:
lol nice - I was just typing that in my terminal. you're quick @scott-8!

dan-v, 2020-05-10:
i'll make that change and push a new release; i was hoping the last duct tape fix would hold a bit longer than this, but apparently google really doesn't want me to parse that page. 

anyways, my plan is to just move to a manually updated file with all of this information updated each month (https://github.com/dan-v/rattlesnakeos-stack/issues/153). i really wanted it to be auto updating, but without a reliable method for getting this information it's more trouble than it's worth.

scott-8, 2020-05-10:
Yeah it would be great if it were auto updating, but at this point it takes way more time just to fix it each month.

laamalif, 2020-05-10:
ERROR: Unable to get latest F-Droid version details. Stopping build.

laamalif, 2020-05-10:
RattlesnakeOS Build FAILED
  Device: crosshatch
  Stack Name: rattlesnakeos-crosshatch-mtumsdsdnkj
  Stack Version: 10.0.7
  Stack Region: us-east-1
  Release Channel: crosshatch-stable
  Instance Type: c5.18xlarge
  Instance Region: us-east-2
  Instance IP: 18.221.229.28
  Build Date: 2020.05.10.10
  Elapsed Time: 0hrs 0min 3sec
  AOSP Build:
  AOSP Vendor Build:
  AOSP Branch:
  Chromium Version: 81.0.4044.138
  F-Droid Version:
  F-Droid Priv Extension Version:
  Build Reason:
 been saved in /etc/ssh/ssh_host_ed25519_key.
Your public key has been saved in /etc/ssh/ssh_host_ed25519_key.pub.
The key fingerprint is:
SHA256:TFe4FrL60EOqauTnNggR2BghSe+QmPr92EN1eJBRJpA root@ip-172-31-6-213
The key's randomart image is:
+--[ED25519 256]--+
|*B    .oo+o..    |
|B.+   E +oo.     |
|o+ .    .=.o     |
|o o    o=.=      |
|.. .   *S+       |
|....  = o        |
| +...o o .       |
|  + =+. .        |
| ..=o.o.         |
+----[SHA256]-----+
Cloud-init v. 18.4-0ubuntu1~18.04.1 running 'modules:config' at Sun, 10 May 2020 10:26:18 +0000. Up 21.58 seconds.
Hit:1 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic InRelease
Get:2 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:3 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:4 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/multiverse Sources [181 kB]
Get:5 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/restricted Sources [5324 B]
Get:6 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/main Sources [829 kB]
Get:7 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/universe Sources [9051 kB]
Get:8 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:9 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [8570 kB]
Get:10 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/universe Translation-en [4941 kB]
Get:11 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [151 kB]
Get:12 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic/multiverse Translation-en [108 kB]
Get:13 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/main Sources [317 kB]
Get:14 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/restricted Sources [7376 B]
Get:15 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/multiverse Sources [6684 B]
Get:16 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/universe Sources [281 kB]
Get:17 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [932 kB]
Get:18 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/main Translation-en [318 kB]
Get:19 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [50.1 kB]
Get:20 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/restricted Translation-en [12.6 kB]
Get:21 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1068 kB]
Get:22 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/universe Translation-en [332 kB]
Get:23 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [15.5 kB]
Get:24 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-updates/multiverse Translation-en [6352 B]
Get:25 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/main Sources [4136 B]
Get:26 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/universe Sources [3452 B]
Get:27 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [7516 B]
Get:28 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/main Translation-en [4764 B]
Get:29 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/universe amd64 Packages [7484 B]
Get:30 http://us-east-2.ec2.archive.ubuntu.com/ubuntu bionic-backports/universe Translation-en [4436 B]
Get:31 http://security.ubuntu.com/ubuntu bionic-security/main Sources [148 kB]
Get:32 http://security.ubuntu.com/ubuntu bionic-security/restricted Sources [5388 B]
Get:33 http://security.ubuntu.com/ubuntu bionic-security/multiverse Sources [3180 B]
Get:34 http://security.ubuntu.com/ubuntu bionic-security/universe Sources [168 kB]
Get:35 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [707 kB]
[... 32 lines cut ...]
2020-05-10 10:26:36: Running aws_notify_simple
==================================
{
    "MessageId": "09e94636-d2af-5f18-b36e-11070bb28233"
}
==================================
2020-05-10 10:26:37: Running aws_logging
==================================
Filesystem      Size  Used Avail Use% Mounted on
udev             69G     0   69G   0% /dev
tmpfs            14G  980K   14G   1% /run
/dev/nvme0n1p1  194G  1.4G  193G   1% /
tmpfs            69G     0   69G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            69G     0   69G   0% /sys/fs/cgroup
/dev/loop0       91M   91M     0 100% /snap/core/6350
/dev/loop1       18M   18M     0 100% /snap/amazon-ssm-agent/930
du: cannot access '/home/ubuntu/rattlesnake-os': No such file or directory
0       total
 10:26:37 up 0 min,  0 users,  load average: 0.33, 0.08, 0.03
Completed 23.1 KiB/23.1 KiB (152.1 KiB/s) with 1 file(s) remaining
upload: var/log/cloud-init-output.log to s3://rattlesnakeos-crosshatch-mtumhnc4hq-logs/crosshatch/1589106397
==================================
2020-05-10 10:26:37: Running aws_notify
==================================

dan-v, 2020-05-10:
@laamalif - I haven't pushed a new release yet. I also found out the android-prepare-vendor was affected by this and have pushed a fix for that (https://github.com/RattlesnakeOS/android-prepare-vendor/commit/0459797b5333803956e3f6d93449f60397affba6) as well. Once I get a successful build, I'll push a new release. I'll re-open this for now for visibility.

scott-8, 2020-05-10:
@dan-v was just about to comment this! I'll let you know how my build goes.

dan-v, 2020-05-10:
@scott-8 - i just finished my build successfully, but let me know if you bump into anything. i pushed a new release now.

scott-8, 2020-05-10:
I had `repo init` fail once but I tried again and it was successful.

dan-v, 2020-05-10:
@scott-8 - any chance you have the error it threw on repo init? could put that in a retry, but not sure if it retries nicely if it did a partial init or something.

scott-8, 2020-05-10:
@dan-v here's the email, let me know if you want full log

```
RattlesnakeOS Build FAILED
  Device: sargo
  Stack Name:
  Stack Version: 10.0.7
  Stack Region: us-east-1
  Release Channel: sargo-stable
  Instance Type: c5.4xlarge
  Instance Region: us-east-2
  Instance IP: 
  Build Date: 2020.05.10.16
  Elapsed Time: 0hrs 3min 26sec
  AOSP Build: QQ2A.200501.001.B2
  AOSP Vendor Build: QQ2A.200501.001.B2
  AOSP Branch: android-10.0.0_r35
  Chromium Version: 81.0.4044.138
  F-Droid Version: 1.8-alpha2
  F-Droid Priv Extension Version: 0.2.11
  Build Reason: 'Stack version 10.0.6 != 10.0.7' 'AOSP build QQ2A.200405.005 != QQ2A.200501.001.B2' 'Chromium version 81.0.4044.138 built but not installed'
======================         ] 77% Unzipping... android-8.1.0/skins/W
[==============================         ] 77% Unzipping... android-8.1.0/skins/W
[==============================         ] 77% Unzipping... android-8.1.0/skins/W
[==============================         ] 77% Unzipping... android-8.1.0/skins/W
.
.
.
.
.
[==============================         ] 77% Unzipping... android-8.1.0/build.p
[==============================         ] 78% Unzipping... android-8.1.0/build.p
[==============================         ] 78% Unzipping... android-8.1.0/build.p
[=======================================] 100% Unzipping... android-8.1.0/build.
[=======================================] 100% Unzipping... android-8.1.0/build.
[=======================================] 100% Unzipping... android-8.1.0/build.
[=======================================] 100% Unzipping... android-8.1.0/build.
==================================
2020-05-10 16:26:04: Running check_chromium
==================================
2020-05-10 16:26:04: Chromium current: 81.0.4044.138
2020-05-10 16:26:04: Chromium latest: 81.0.4044.138
2020-05-10 16:26:04: Chromium latest (81.0.4044.138) matches current (81.0.4044.138)
==================================
2020-05-10 16:26:04: Running aosp_repo_init
==================================
gpg: keybox '/home/ubuntu/.repoconfig/gnupg/pubring.kbx' created
gpg: /home/ubuntu/.repoconfig/gnupg/trustdb.gpg: trustdb created
gpg: key 16530D5E920F5C65: public key "Repo Maintainer <repo@android.kernel.org>" imported
gpg: key 67B7E448692B382C: public key "Conley Owens <cco3@android.com>" imported
gpg: Total number processed: 2
gpg:               imported: 2
Get https://gerrit.googlesource.com/git-repo/clone.bundle
Get https://gerrit.googlesource.com/git-repo
fatal: unable to access 'https://gerrit.googlesource.com/git-repo/': gnutls_handshake() failed: The TLS connection was non-properly terminated.

==================================
2020-05-10 16:26:16: Running aosp_repo_modifications
==================================
==================================
2020-05-10 16:26:16: Running aosp_repo_sync
==================================
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
error: command 'sync' requires repo to be installed first.
         Use "repo init" to install it here.
==================================
2020-05-10 16:26:16: Running aws_logging
==================================
Filesystem      Size  Used Avail Use% Mounted on
udev             16G     0   16G   0% /dev
tmpfs           3.1G  760K  3.1G   1% /run
/dev/nvme0n1p1  194G  4.1G  190G   3% /
tmpfs            16G     0   16G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            16G     0   16G   0% /sys/fs/cgroup
/dev/loop0       91M   91M     0 100% /snap/core/6350
/dev/loop1       18M   18M     0 100% /snap/amazon-ssm-agent/930
tmpfs            20M     0   20M   0% /home/ubuntu/rattlesnake-os/keys
16K     /home/ubuntu/rattlesnake-os
16K     total
 16:26:16 up 4 min,  0 users,  load average: 3.10, 1.17, 0.45
Completed 256.0 KiB/5.0 MiB (1.8 MiB/s) with 1 file(s) remaining
Completed 512.0 KiB/5.0 MiB (3.6 MiB/s) with 1 file(s) remaining
Completed 768.0 KiB/5.0 MiB (5.0 MiB/s) with 1 file(s) remaining
Completed 1.0 MiB/5.0 MiB (6.7 MiB/s) with 1 file(s) remaining 
Completed 1.2 MiB/5.0 MiB (8.3 MiB/s) with 1 file(s) remaining 
.
.
.
```

dan-v, 2020-05-10:
thanks @scott-8 - that error definitely looks retry-able. i added a retry (https://github.com/dan-v/rattlesnakeos-stack/commit/4eba30a9db8a9ca9c366ae2bb73cd6350776b992) and i'll push it in future release. there's probably a bunch of additional places where retries would be beneficial, but not going to tackle that right now.

scott-8, 2020-05-10:
This is the first time I've found anything that could be fixed with a retry, so it's probably safe to just wait any see if anything like that happens again

closed 2020-05-13 by dan-v