dan-v/rattlesnakeos-stack #161: `instance-regions` should respect `region` config variable

ddxor, 2020-05-24:
**Is your feature request related to a problem? Please describe.**
I created a key in `eu-west-1` and set `eu-west-1` as my region in `config`. I tried to run a build, and it failed because my instance was spun up in a `us-west-` region.

**Describe the solution you'd like**
There's a few options-

1. Make it explicitly clear in the documentation that keys must be uploaded to each region specified in `instance-regions`, including the default ones.
2. Enforce the uploading of keys to each region specified in `instance-regions` (or defaults), perhaps when running `deploy`.
3. Replace the `instance-regions` default value with the `region` specified at `config` time.

dan-v, 2020-05-24:
Hi there @ddxor. I do have the step listed in the `Prerequisites` section to generate a key in all regions, so let me know if there is a way to make that more clear. I never got around to automating that step completely, but I think it makes sense to streamline this a bit to avoid confusion.

ddxor, 2020-05-24:
I probably skimmed the `Prerequirites` section in all honesty. I will look to work up a solution, maybe a check at the `deploy` stage. I will put up a PR when I have something to show.

dan-v, 2020-05-24:
@ddxor - would love a PR for a check in the deploy stage. Thanks!

closed 2022-03-05 by dan-v