dan-v/rattlesnakeos-stack #165: Pixel 2 (walleye) bootloop with microG

Lacxp, 2020-08-24:
### Prerequisites

* [X] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [X] I am able to reproduce this issue without any advanced customization options (since I put only one patched playstore in system-privapp)

### Description

Hi guys,
Yesterday I got myself finally to make ROS for a Pixel 2 (walleye). I used the dan-v microg repo for sign. spoofing and I made my own microG repo for prebuilts because I needed a tiny modification with NanoDroid patched playstore.  First time, the build failed. It was because even I did not put in "FakeStore" within the repo's modules in ROS config file, it got installed and thereby patched playstore couldn't. So I only left patched playstore in my repo. The built was succesful, but I got bootloop. During flashing everything seems fine.

**Expected behavior:** [What you expected to happen]
 Since I put only one patched playstore in system-privapp by referencing to my own repo, I expected it to at least boot up.

**Actual behavior:** [What actually happened]
Bootloop.

### Platform
Windows

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-instance-type = "t3.nano"
attestation-max-price = ".005"
attestation-server = false
chromium-version = ""
device = "walleye"
email = "****"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.9xlarge"
max-price = "1.8"
name = "rattlesnakeos-walleye-**********"
region = "us-west-2"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.0"
ssh-key = "rattlesnakeos"

[[custom-patches]]
  repo = "https://github.com/RattlesnakeOS/community_patches"
  patches = ["00001-global-internet-permission-toggle.patch", "00008-round-icon.patch"]
	
[[custom-patches]]
  patches = ["00002-microg-sigspoof.patch"]
  repo = "https://github.com/RattlesnakeOS/microg"
	
[[custom-prebuilts]]
  modules = ["AuroraServices", "GmsCore", "GsfProxy", "PhoneSky(NanoDroid)", "com.google.android.maps"]
  repo = "https://notabug.org/Lacxp/and_pre_apks.git"
	
```

### Carrier
e.g. Telenor HU

### Email Notification Output
```
RattlesnakeOS Build SUCCESS
  Device: walleye
  Stack Name: rattlesnakeos-walleye-ihs3ar2ern
  Stack Version: 10.0.9
  Stack Region: us-west-2
  Release Channel: walleye-stable
  Instance Type: c5.9xlarge
  Instance Region: us-east-2
  Instance IP: 18.224.179.117
  Build Date: 2020.08.23.19
  Elapsed Time: 3hrs 42min 25sec
```

### Full log
I kindly share it, but it seems pretty long, so I attached it.

[1598222606.zip](https://github.com/dan-v/rattlesnakeos-stack/files/5116705/1598222606.zip)

Thanks, and keep up with this beast project ;)

dan-v, 2020-08-28:
Hi @Lacxp! I don't have a walleye device to test against, but I'd definitely recommend building without any patches as a starting point and then add back patches one by one. Unfortunately right now there really isn't a great way to build and test out custom patches in a sane way and the turn around time for testing is a full build which takes a ton of time. Also being able to create a userdebug build might help in this case as you might at least get some logcat output on boot that would tell you what the issue is. I've filed an enhancement for that (https://github.com/dan-v/rattlesnakeos-stack/issues/166).

Lacxp, 2020-08-28:
Thanks so much Dan! When I'll have time I work it out 😉

dan-v, 2020-10-03:
Gonna close this out, feel free to re-open if you're able to track this down to an issue without custom patches.

closed 2020-10-03 by dan-v