dan-v/rattlesnakeos-stack #168: OTA s3 uploads fail after build

david-collett, 2020-09-07:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options

### Description

I am consistently finding that OTA files are not appearing on s3 following a build (but target and factory files do).
At first I thought that perhaps rattlesnake-stack does not produce them for initial/manual builds, but on looking at the logs it seems that they are but the s3 uploads is failing. Do you have any ideas on why that may be?

Please see build log output below.

### Steps to Reproduce

1. start build

**Expected behavior:**

I assumed the build would produce signed OTA files in addition to factory images.

**Actual behavior:**

Only factory images are produced.

### Full log
```
sargo-factory-2020.09.07.07.tar -> 16/21 threads: [0 2 3 5 1 11 6 4 12 10 8 13 14 15 7 9 4 2 3 1 0 ] 4146995200 -> 1464680192 35.319%

real	3m49.360s
user	36m20.846s
sys	0m18.711s
==================================
2020-09-07 09:28:42: Running aws_upload
==================================
--2020-09-07 09:28:42--  https://rattlesnakeos-sargo-0xgdw2uvhb-release.s3.amazonaws.com/sargo-stable
Resolving rattlesnakeos-sargo-0xgdw2uvhb-release.s3.amazonaws.com (rattlesnakeos-sargo-0xgdw2uvhb-release.s3.amazonaws.com)... 52.219.113.10
Connecting to rattlesnakeos-sargo-0xgdw2uvhb-release.s3.amazonaws.com (rattlesnakeos-sargo-0xgdw2uvhb-release.s3.amazonaws.com)|52.219.113.10|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-09-07 09:28:42 ERROR 403: Forbidden.

upload failed: release-sargo-2020.09.07.07/sargo-ota_update-2020.09.07.07.zip to s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-ota_update-2020.09.07.07.zip An error occurred (AccessDenied) when calling the CreateMultipartUpload operation: Access Denied
upload: release-sargo-2020.09.07.07/sargo-factory-2020.09.07.07.tar.xz to s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-factory-latest.tar.xz
==================================
2020-09-07 09:28:55: Running cleanup_target_files
==================================
download: s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-target/sargo-target-files-2020.08.27.04.zip to ../sargo-target/sargo-target-files-2020.08.27.04.zip
delete: s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-target/sargo-target-files-2020.08.27.04.zip
upload: ../out/release-sargo-2020.09.07.07/sargo-target_files-2020.09.07.07.zip to s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-target/sargo-target-files-2020.09.07.07.zip
==================================
2020-09-07 09:30:08: Running checkpoint_versions
==================================
upload failed: - to s3://rattlesnakeos-sargo-0xgdw2uvhb-release/sargo-vendor An error occurred (AccessDenied) when calling the PutObject operation: Access Denied
==================================
2020-09-07 09:30:11: Running aws_notify
==================================
{
    "MessageId": "4e4403e7-e5db-5b0b-8fa2-54281a3fce99"
}
```

dan-v, 2020-09-07:
Thanks for reporting this issue @david-collett. This is definitely a strange one; I have never seen this issue before and I'm not able to reproduce. Could you share what region you have setup your stack in? It appears to be failing on any of the s3 copy commands that are configuring public ACLs. Could you double check that you don't have any 'Block public access' settings enabled here: https://s3.console.aws.amazon.com/s3/buckets/rattlesnakeos-sargo-0xgdw2uvhb-release/?region=us-west-2&tab=permissions.

david-collett, 2020-09-08:
Hi Dan,
Nothing is blocked in that link specifically (i.e. the release bucket).

However I found that all public access is blocked at the account level ("Block public access (account settings)" on the left)!
That would explain it! and also why I can see update checks are failing in adb logcat.
What should these (account-wide) settings be? Can I assume if I turn off block public access, buckets are still only accessible by ACL's and everything isn't made public (inc. keys)?

Dave

dan-v, 2020-09-08:
@david-collett - that's quite interesting! How new is your AWS account? I'm wondering if for new accounts they have it default to block all public access now? You should be fine flipping that global setting off; I have only enabled public access on very specific files in the stack, namely the OTA update files and the metadata files used for update checking. I'll need to read up a bit more on the global S3 settings that block public access and see if there is some more elegant way to workaround this than disabling the entire feature.

david-collett, 2020-09-08:
The account is not new, but is it's shared with some other developers so perhaps it was turned off manually at some point.
I can confirm that builds are working now and producing the ota zip file in s3.
Thanks for your help.

dan-v, 2020-09-08:
No worries - glad you got it worked out.

closed 2020-09-08 by dan-v