dan-v/rattlesnakeos-stack #172: Authenticate with git+googlesource.com to avoid rate limiting

ddxor, 2020-09-22:
**Is your feature request related to a problem? Please describe.**
I found various entries in my `/var/log/clout-init-output.log` related to hitting rate limiting on googlesource.com. Specifically, I hit limits at the `aosp_repo_sync` method. E.g:

`error: RPC failed; HTTP 429 curl 22 The requested URL returned error: 429 Too Many Requests
fatal: The remote end hung up unexpectedly`

**Describe the solution you'd like**
Provide some method for a user to avoid hitting rate limits on googlesource.com. Perhaps via a `.gitcookies` file with authentication data retrieved from 'Generate Password' part of [the relevant go.googlesource.com page](https://go.googlesource.com/)

This could include documentation, and some method for inserting the `.gitcookies` data. Perhaps stored in temporary storage like keys. TBC.

**Describe alternatives you've considered**
Connections will be retried eventually, as I found out, so another option is do nothing and accept that build times will be delayed by this timeout/retry cycle.

**Additional context**
I will work on this..

dan-v, 2020-09-25:
@ddxor - I believe I attempted the gitcookies approach before, and ended up getting rate limited even worse. The only real approach to avoid the rate limiting would be to reduce the sync threads, which maybe there is a sweet spot that avoids 429s that might slightly improve download time. I think the last time I looked at timing it took about 10 minutes to sync, which seems fairly quick considering the size of AOSP repos. Anyways, I'd be happy to take a PR for something that reduces 429s and either improves sync times or doesn't negatively impact it too much.

ddxor, 2020-09-25:
Funnily enough I only ran into this issue when I used a `c5a.16xlarge` instance, so it's quite possible that this only becomes an issue if you have a beefier server without enough network/IO to trigger a 429. (warning: anecdotal information!)

I will look into this further.

dan-v, 2021-03-01:
Closing this out. Feel free to re-open if this does help with rate limiting.

closed 2021-03-01 by dan-v