dan-v/rattlesnakeos-stack #173: Build not started

MalteKiefer, 2020-10-12:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options

### Description

Hi,

I tried to start an instance with this config:

```toml
chromium-version = ""
device = "blueline"
email = "my mail"
encrypted-keys = false
hosts-file = "https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts"
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.18xlarge"
max-price = "2.00"
skip-price = "2.00"
name = "rattlesnakeos-blueline-efy80u1984"
region = "us-east-1"
ssh-key = "rs"
[[custom-patches]]
  patches = [
        "00001-global-internet-permission-toggle.patch",
        "00002-round-icon.patch",
        "00003-Enable-WiFi-and-4G-calling-sargo-bonito.patch",
  ]
  repo = "https://github.com/RattlesnakeOS/community_patches"
  branch = "11.0"
```

I have done this steps:

```
/usr/bin/rattlesnakeos-stack deploy
#
/usr/bin/rattlesnakeos-stack build start
```

I have got this mail:

```
Successfully requested a spot instance.

Stack Name: rattlesnakeos-blueline-efy80u1984
Device: blueline
Force Build: False
Instance Type: c5.18xlarge
Cheapest Region: us-east-2
Cheapest Hourly Price: $0.862500
```

This was for 8 hours. Since then nothing happened.

```
➜  ~ rattlesnakeos-stack build list
INFO[0000] Using config file: /home/malte/.rattlesnakeos.toml
INFO[0000] Looking for builds for stack rattlesnakeos-blueline-efy80u1984 in the following regions: us-west-2,us-west-1,us-east-2
INFO[0003] No active builds found
```

dan-v, 2020-10-13:
@MalteKiefer - it is possible there was no spot instance availability. If you go into the EC2 AWS console after starting a build, you can go to Spot Intances page and you should be able to see the status of your spot request. Generally spot instance requests get immediately fulfilled, but depending on how much capacity AWS has available for a certain instance type in a certain region or AZ, it may not be fulfilled. The current code sets a max 12 hour window on a spot request, so likely your request is expired at this point. Anyways, my recommendations 1) check AWS spot instance status in console after starting a build 2) pick a different instance type.

MalteKiefer, 2020-10-13:
Thanks.
Changed the type and it started immediately.

closed 2020-10-13 by MalteKiefer