dan-v/rattlesnakeos-stack #174: 2-button system navigation no longer available with 11.0 build

khimaros, 2020-10-12:
### Prerequisites

* [ ] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack

testing this with 11.0.3 now, this report is for a release built with 11.0.1

* [x] I am able to reproduce this issue without any advanced customization options

please let me know if you need me to test without the seedvault bits

### Description

unable to use 2-button system navigation with latest builds. Google builds for blueline still seem to include.

### Steps to Reproduce

1. Settings > System > Gestures
2. Select "System navigation"
3. Look for "2-button navigation" option

**Expected behavior:**

a "2-button navigation" option should be present and selectable

**Actual behavior:**

only "Gesture navigation" and "3-button navigation" are available

### Platform
Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
attestation-server = false
chromium-version = ""
device = "blueline"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "rattlesnakeos-blueline-<masked>"
region = "us-west-2"
schedule = "rate(14 days)"
skip-price = "0.68"
ssh-key = "rattlesnakeos"

### Seedvault

[[custom-manifest-remotes]]
  name = "github-stevesoltys"
  fetch = "https://github.com/stevesoltys/"
  revision = "master"

[[custom-manifest-projects]]
  path = "packages/apps/Seedvault"
  name = "seedvault-prebuilt"
  remote = "github-stevesoltys"
  modules = [ "Seedvault" ]
```

### Carrier
T-Mobile

### Email Notification Output
Should look something like this. You can mask stack name from here and you may want to verify contents of log output before pasting.
```
RattlesnakeOS Build SUCCESS
  Device: blueline
  Stack Name: rattlesnakeos-blueline-<masked>
  Stack Version: 11.0.1 
  Stack Region: us-west-2
  Release Channel: blueline-stable
  Instance Type: c5.4xlarge
  Instance Region: us-east-2
  Instance IP: <masked>
  Build Date: 2020.10.06.19
  Elapsed Time: 6hrs 58min 18sec
  AOSP Build: RP1A.201005.004
  AOSP Vendor Build: RP1A.201005.004
  AOSP Branch: android-11.0.0_r4
  Chromium Version: 86.0.4240.68
  F-Droid Version: 1.9
  F-Droid Priv Extension Version: 0.2.11
  Build Reason: 'Stack version 10.0.10 != 11.0.1' 'AOSP build QQ3A.200805.001 != RP1A.201005.004' 'Chromium version 84.0.4147.111 != 86.0.4240.68'
```

dan-v, 2020-10-13:
@khimaros - interesting, I don't use the 2 button nav so I didn't realize. Looks like they removed it from being enabled by default here: https://android.googlesource.com/platform/frameworks/base/+/2aa347798732f8ac1a87f0316af7bc345f2c184a.  I'll think about whether I want to pull this in to default build.

khimaros, 2020-10-13:
@dan-v thank you for the reply and digging into the root cause. is there a recommended way for users to add to LOCAL_REQUIRED_MODULES with the ROS config?

dan-v, 2020-10-17:
@khimaros - there was a community patch contribution by @losuler for adding 2 button navigation back: https://github.com/RattlesnakeOS/community_patches/tree/11.0

e.g. 
```
[[custom-patches]]
  patches = [
    "00005-2-button-navigation.patch",
  ]
  repo = "https://github.com/rattlesnakeos/community_patches"
  branch = "11.0"
```

closed 2020-10-17 by dan-v