dan-v/rattlesnakeos-stack #180: Support for Pixel 4a 5G Bramble

coxtor, 2021-01-25:
**Is your feature request related to a problem? Please describe.**
The Pixel 4a 5G is not supported.
**Describe the solution you'd like**

Support for Pixel 4a 5G, Bramble.

**Additional context**
After support for Pixel 5, Redfin, has been merged #178 is support for the 4a 5G possible now ?

dan-v, 2021-01-30:
@coxtor - there is support for Pixel 4a 5G in android-prepare-vendor now (https://github.com/AOSPAlliance/android-prepare-vendor/commit/4e3aea63bd5b16b409cce34538b218b186654f5e). Happy to take a PR from anyone on adding the boilerplate code/templates updates to rattlesnakeos-stack, otherwise I'll probably add support for it myself in the near future.

coxtor, 2021-02-27:
I'd be glad to help, however I am not familiar with the process - is it sufficient to replicate the changes #178 in and adjust the lines accordingly for bramble?

dan-v, 2021-02-27:
> I'd be glad to help, however I am not familiar with the process - is it sufficient to replicate the changes #178 in and adjust the lines accordingly for bramble?

Hi @coxtor! Indeed, it is sufficient to just replicate those changes. The only other change is adding the device to the [latest](https://github.com/RattlesnakeOS/latest/blob/11.0/latest.json) repo. I'd hold off though as I have a pretty large set of changes coming soon, and I have already added support for Pixel 4a 5G as part of those changes.

PR #182 (2021-03-01): refactoring and changes to how customizations work
* new: add support for custom config repos and deprecate old customization methods (custom-patches, custom-manifest-remotes, custom-manifest-projects, custom-prebuilts)
* new: also add support for core config repos that can be used to build an OS with plain AOSP rather than using the default RattlesnakeOS core. closes #129.
* new: add support for Pixel 4a 5G. closes #180.
* deprecate: encrypted-keys option (for now) to reduce build script complexity
* deprecate: hosts-file (can be added in custom config repo)
* move version checking to lambda  [... cut]
commit 61be4195 (2021-03-01): refactoring and changes to how customizations work (#182)

* new: add support for custom config repos and deprecate old customization methods (custom-patches, custom-manifest-remotes, custom-manifest-projects, custom-prebuilts)
* new: also add support for core config repos that can be used to build an OS with plain AOSP rather than using the default RattlesnakeOS core. closes #129.
* new: add support for Pixel 4a 5G. closes #180.
* deprecate: encrypted-keys option (for now) to reduce build script complexity
* deprecate: hosts-file (can be added in custom config repo)
* move version checking to [... cut]
closed 2021-03-01 by dan-v