dan-v/rattlesnakeos-stack #185: Error deploying on Windows

scott-8, 2021-03-10:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any customization options

### Description

After switching to 11.0.8, I get an error when trying to deploy on Windows.

```bash
time="2021-03-09T23:33:16-05:00" level=info msg="Executing terraform apply for stack rattlesnakeos-xxxxxxx"
There are some problems with the configuration, described below.

The Terraform configuration must be valid before initialization so that
Terraform can determine which modules and providers need to be installed.

Error: Error parsing D:\Test\output_rattlesnakeos-xxxxxxx\main.tf: At 37:26: illegal char escape

time="2021-03-09T23:33:16-05:00" level=fatal msg="exit status 1"
```

The line in question in `main.tf` is `   default     = "D:\Code\output_rattlesnakeos-xxxxxxxx\build.sh"`

Doing a dry run works fine.

### Steps to Reproduce

1. Use Windows.
2. Deploy stack.

**Expected behavior:** Successful deployment. 

**Actual behavior:** Error.

### Platform
Windows

### Full Config Used for Setup
You can mask stack name and email from here.
```
chromium-version = ""
device = "crosshatch"
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "<not important>"
region = "us-west-2"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.00"
ssh-key = "rattlesnakeos"
```

### Carrier
N/A

### Email Notification Output
N/A

### Full log
N/A

scott-8, 2021-03-10:
@dan-v can we reopen this? I'm getting this in the build log, presumable because newlines in Windows are `\r\n`. Not sure where the best place to fix this would be. I manually changed the newlines in `build.sh` to `\n` and it got past this part.

```log
Setting up python3-botocore (1.16.19+repack-1ubuntu0.20.04.1) ...
Setting up python3-s3transfer (0.3.3-1) ...
Setting up awscli (1.18.69-1ubuntu0.20.04.1) ...
Completed 25.2 KiB/25.2 KiB (221.3 KiB/s) with 1 file(s) remaining
download: s3://rattlesnakeos-xxxxxxxxxxx-script/build.sh to home/ubuntu/build.sh
/home/ubuntu/build.sh: line 2: $'\r': command not found
RELEASE=20200301

AOSP_BUILD_ID=RQ2A.210305.006

AOSP_TAG=android-11.0.0_r32

CHROMIUM_VERSION=88.0.4324.181

LOCAL_MANIFEST_REVISIONS=................

/home/ubuntu/build.sh: line 16: $'\r': command not found
/home/ubuntu/build.sh: line 32: $'\r': command not found
/home/ubuntu/build.sh: line 40: $'\r': command not found
/home/ubuntu/build.sh: line 41: syntax error near unexpected token `$'{\r''
/home/ubuntu/build.sh: line 41: `import_keys() {
'
Cloud-init v. 20.4.1-0ubuntu1~20.04.1 running 'modules:final' at Wed, 10 Mar 2021 05:42:51 +0000. Up 21.04 seconds.
2021-03-10 05:43:18,717 - cc_scripts_user.py[WARNING]: Failed to run module scripts-user (scripts in /var/lib/cloud/instance/scripts)
2021-03-10 05:43:18,717 - util.py[WARNING]: Running module scripts-user (<module 'cloudinit.config.cc_scripts_user' from '/usr/lib/python3/dist-packages/cloudinit/config/cc_scripts_user.py'>) failed
Cloud-init v. 20.4.1-0ubuntu1~20.04.1 finished at Wed, 10 Mar 2021 05:43:18 +0000. Datasource DataSourceEc2Local.  Up 47.26 seconds
```

dan-v, 2021-03-10:
Reopened. Ah, did not consider this. Sorry I don't have a Windows box, so mostly just hoping it worked. I'll spin up a Windows host on AWS when I get a chance and work through the issues. Happy to take more PRs as well though.

PR #186 (2021-03-10): Fix Windows file paths
Fixes #185 

I tested this fix out and everything deployed okay (minus some warnings, but that could be because I built it on Windows and the makefile wasn't happy)
PR #187 (2021-03-11): Fix for Windows deployment
A small change that removes the carriage returns from the build script. There are still carriage returns in `main.tf` and `lambda_spot_function.py` but it seems like those are handled fine.

I'm running a test build right now but everything looks good so far. Hopefully fixes #185.
closed 2021-03-11 by dan-v