dan-v/rattlesnakeos-stack #189: Signature doesn't match any trusted key

losuler, 2021-04-21:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [ ] I am able to reproduce this issue without any customization options

### Description

Updating from stack version 11.0.6 (which built RQ1A.210205.004, the February security update) to 11.0.10 (which built RQ2A.210405.005, the April security update) and completing the migration steps as per https://old.reddit.com/r/RattlesnakeOS/comments/lvuwfi/mar_2021_update_and_v1107/. Results in the RattlesnakeOS Updater showing the error, "Signature doesn't match any trusted key" upon downloading the update.

I've included my `~/.rattlesnakeos.toml` that I used for this and prior to the migration further below. My only guess is this is possibly related the fact that I was using `encrypted-keys = true` prior to the migration. Could this be the reason for the update error?

### Steps to Reproduce

1. Downloaded the 11.0.10 stack binary.
2. Removed custom-patches, hosts-file, encrypted-keys and ignore-version-check from `~/.rattlesnakeos.toml`.
3. Ran the command `./rattlesnakeos-stack deploy --dry-run --save-config`.
4. Created a custom config repo, updated the remote fetch URL and added patches I was using previously. See [losuler/custom-config-repo](https://github.com/losuler/custom-config-repo).
5. Added the custom config repo to `~/.rattlesnakeos.toml`.
6. Ran `./rattlesnakeos-stack deploy` followed by `./rattlesnakeos-stack build start`.
7. Build success for RQ2A.210405.005.
8. Tap "Check for updates" in RattlesnakeOS Updater, wait for download to finish.
9. Notification shows error details, "Signature doesn't match any trusted key".

### Platform
Linux.

### Full Config Used for Setup
You can mask stack name and email from here.
```
chromium-build-disabled = false
chromium-version = ""
cloud = "aws"
core-config-repo = "https://github.com/rattlesnakeos/core-config-repo"
core-config-repo-branch = "11.0"
custom-config-repo = "https://github.com/losuler/custom-config-repo"
custom-config-repo-branch = "11.0"
device = "sunfish"
email = "[redacted]"
instance-regions = "ap-northeast-1"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "[redacted]"
output-dir = ""
region = "ap-southeast-1"
releases-url = "https://raw.githubusercontent.com/RattlesnakeOS/releases/11.0/latest.json"
schedule = ""
skip-price = "1.00"
ssh-key = "rattlesnakeos"
```

### Full Config Used for Setup Prior
You can mask stack name and email from here.
```
chromium-version = ""
device = "sunfish"
email = "[redacted]"
encrypted-keys = true
hosts-file = ""
ignore-version-checks = false
instance-regions = "ap-northeast-1"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "[redacted]"
region = "ap-southeast-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "1.00"
ssh-key = "rattlesnakeos"

[[custom-patches]]
  patches = [
    "00001-global-internet-permission-toggle.patch",
    "00002-round-icon.patch",
    "00003-enable-volte-wifi-calling.patch",
    "00004-use-cloudflare-dns.patch",
    "00005-2-button-navigation.patch"
  ]
  repo = "https://github.com/losuler/community_patches"
  branch = "11.0"
```

### Error Message
[image]

dan-v, 2021-04-21:
hi @losuler. This is indeed because you were using the encrypted keys option which is no longer currently supported (still trying to figure out what i want to do with that, but feel free to add your +1 in https://github.com/dan-v/rattlesnakeos-stack/issues/188). Basically new keys got generated in `<stackname-keys>` which do not match what you are currently using and the OTA update is being rejected. In order to go from encrypted keys option back to non encrypted, you'll need to:
* grab all your encrypted keys from the `<stackname-keys-encrypted>` S3 bucket
* decrypt them (you can see how to do that here https://github.com/dan-v/rattlesnakeos-stack/blob/8839b54dfea7183b4893ca5de2e0016b72deccc6/templates/build_template.go#L408)
* delete any existing keys in `<stackname-keys>`
* upload the non encrypted keys to `<stackname-keys>`
* do another build which will use the correct keys as you are currently running on your phone

Let me know if that is enough detail. Apologies for dropping support for the encrypt keys option, I think there is a total of 4 people that I've heard from so far that were using this setup, so definitely still some interest in bringing this back in some form.

losuler, 2021-04-22:
Thanks for the help. After I wrote this issue I was looking at the S3 bucket and figured this probably was the case seeing as `stackname-keys` had newly generated keys in them. I'm glad it was just as easy as replacing those files with my decrypted ones. After doing this the rebuild and OTA update applied without error.

Thanks again for all your work on RattlesnakeOS. I really appreciate it.

closed 2021-04-22 by losuler