dan-v/rattlesnakeos-stack #190: Terraform error on deploy after upgrading from 10.0.6

sal-limones, 2021-05-02:
OS: Ubuntu Linux 20.10
Stack version: 11.0.10

ubuntu@laptop:~$ ./rattlesnakeos-stack deploy
INFO[0000] using config file: /home/ubuntu/.rattlesnakeos.toml 
INFO[0000] Current settings:                            
chromium-build-disabled: false
chromium-version: ""
cloud: aws
core-config-repo: https://github.com/rattlesnakeos/core-config-repo
core-config-repo-branch: "11.0"
custom-config-repo: ""
custom-config-repo-branch: ""
device: blueline
email: email_here
instance-regions: us-west-2,us-west-1,us-east-1,us-east-2
instance-type: c5.4xlarge
max-price: "1.00"
name: rattlesnakeos-blueline
output-dir: ""
region: us-east-2
releases-url: https://raw.githubusercontent.com/RattlesnakeOS/releases/11.0/latest.json
schedule: ""
skip-price: "0.68"
ssh-key: rattlesnakeos

Do you want to continue : y
INFO[0000] all generated files will be placed in /home/ubuntu/output_rattlesnakeos-blueline
INFO[0003] Downloading Terraform binary from URL: https://releases.hashicorp.com/terraform/0.11.14/terraform_0.11.14_linux_amd64.zip 
INFO[0004] Rendering all templates files for stack rattlesnakeos-blueline 
INFO[0004] Creating/updating non terraform resources for stack rattlesnakeos-blueline 
INFO[0008] Executing terraform apply for stack rattlesnakeos-blueline

Initializing the backend...

Successfully configured the backend "s3"! Terraform will automatically
use this backend unless the backend configuration changes.

Initializing provider plugins...
- Checking for available provider plugins on https://releases.hashicorp.com...

Error installing provider "aws": openpgp: signature made by unknown entity.

Terraform analyses the configuration and state and automatically downloads
plugins for the providers used. However, when attempting to download this
plugin an unexpected error occured.

This may be caused if for some reason Terraform is unable to reach the
plugin repository. The repository may be unreachable if access is blocked
by a firewall.

If automatic installation is not possible or desirable in your environment,
you may alternatively manually install plugins by downloading a suitable
distribution package and placing the plugin's executable file in the
following directory:
    terraform.d/plugins/linux_amd64

FATA[0013] exit status 1

dan-v, 2021-05-02:
@sal-limones - thanks for the bug report! Terraform changed their signing keys due to [a security incident ](https://discuss.hashicorp.com/t/terraform-updates-for-hcsec-2021-12/23570), so I'll need to make a new release with an updated version of Terraform.

closed 2021-05-02 by dan-v

dan-v, 2021-05-02:
Pushed a new release with this fix @sal-limones: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v11.0.11.

commit 43b74af7 (2021-05-02): fix terraform gpg error

* fix issue with terraform on init that was giving error 'openpgp: signature made by unknown entity'. this is due to hashicorp rotation their signing keys in response to a security incident (https://discuss.hashicorp.com/t/terraform-updates-for-hcsec-2021-12/23570). fix is upgrading from 0.11.14->0.11.15. closes #190.