dan-v/rattlesnakeos-stack #196: Problem with android-prepare-vendor

nullchinchilla, 2021-12-17:
I "brought my own" APV, but am getting this:

```
...
Completed 30.7 MiB/33.0 MiB (53.2 MiB/s) with 1 file(s) remaining
Completed 31.0 MiB/33.0 MiB (53.6 MiB/s) with 1 file(s) remaining
Completed 31.2 MiB/33.0 MiB (53.9 MiB/s) with 1 file(s) remaining
Completed 31.5 MiB/33.0 MiB (54.3 MiB/s) with 1 file(s) remaining
Completed 31.7 MiB/33.0 MiB (54.7 MiB/s) with 1 file(s) remaining
Completed 32.0 MiB/33.0 MiB (55.0 MiB/s) with 1 file(s) remaining
Completed 32.2 MiB/33.0 MiB (55.4 MiB/s) with 1 file(s) remaining
Completed 32.5 MiB/33.0 MiB (55.8 MiB/s) with 1 file(s) remaining
Completed 32.7 MiB/33.0 MiB (56.2 MiB/s) with 1 file(s) remaining
Completed 33.0 MiB/33.0 MiB (55.2 MiB/s) with 1 file(s) remaining
download: s3://rattlesnakeos-blueline-djfhu4y7s0-release/chromium/TrichromeChrome.apk to external/chromium/prebuilt/arm64/TrichromeChrome.apk
==================================
2021-12-17 01:45:11: Running setup_vendor
==================================
2021-12-17 01:45:11: Running /home/ubuntu/core/hooks/setup_vendor_pre.sh
error: bad tree object bde54dfa66e1092893e2c1bfa78385a35588387a
timeout: failed to run command ‘/home/ubuntu/aosp/vendor/android-prepare-vendor/execute-all.sh’: No such file or directory
Filesystem      Size  Used Avail Use% Mounted on
```

What's wrong?

dan-v, 2021-12-17:
It appears the APV you brought may not be valid. Could you share your config related to APV?

nullchinchilla, 2021-12-17:
```
apv-remote="https://github.com/GrapheneOS/android-prepare-vendor.git"
apv-branch="12"
apv-revision="bde54dfa66e1092893e2c1bfa78385a35588387a"
```

dan-v, 2021-12-17:
Looks like your `apv-remote` setting just needs to be updated. I haven't really documented or added any error checking for these APV settings as it was quickly thrown in there as a temporary workaround when GrapheneOS decided to start playing licensing games, but happy to take any PRs to clean this up.
```
apv-remote = "https://github.com/GrapheneOS/"
apv-branch = "12"
apv-revision = "bde54dfa66e1092893e2c1bfa78385a35588387a"
```

nullchinchilla, 2021-12-17:
Thanks! Also, embedding the above lines in the code is very unlikely to be a licensing violation. see `msttcorefonts-installer` and similar things

closed 2022-01-24 by dan-v