dan-v/rattlesnakeos-stack #197: No space left on device - build failure

IamJeffG, 2022-01-09:
### Prerequisites

* [x] I am running 12.0.3
* [ ] I am able to reproduce this issue without any customization options
    - I didn't try this, so feel free to push back.  [My customization are pretty standard, and do include microg](https://github.com/IamJeffG/rattlesnake-custom-config-repo)

### Description

My January build for blueline failed with

    cp: error writing 'tmp/blueline-sp1a.210812.016.a2/image-blueline-sp1a.210812.016.a2.zip': No space left on device

See detailed logs below, which sure enough shows that all 291G on /dev/root/ are used.   I think that volume size [might be set here](https://github.com/dan-v/rattlesnakeos-stack/blob/ddbb571dd2bc8937707f5d435fa6ccecebdfa0e8/templates/lambda.py#L152-L153), I'm curious if others have had to bump this up and if there's a config option to do so.  Or is it outrageous that I'd fill the disk and maybe I should just try the build again?

### Steps to Reproduce

I just waited for the January build to start according to cron.  No change from the December build, which succeeded.

This build started but failed to complete.  None of the files in my S3 release bucket seem changed (which sounds right given that the build failed).

### Platform
Linux

### Full Config Used for Setup
You can mask stack name and email from here.
```
apv-remote = "https://github.com/GrapheneOS/"
apv-branch = "12"
apv-revision = "bde54dfa66e1092893e2c1bfa78385a35588387a"
chromium-build-disabled = false
chromium-version = ""
cloud = "aws"
core-config-repo = "https://github.com/rattlesnakeos/core-config-repo"
core-config-repo-branch = "12.0"
custom-config-repo = "https://github.com/IamJeffG/rattlesnake-custom-config-repo"
custom-config-repo-branch = ""
device = "blueline"
email = *******
instance-regions = "us-west-2,us-west-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = *******
output-dir = ""
region = "us-east-2"
releases-url = "https://raw.githubusercontent.com/RattlesnakeOS/releases/12.0/latest.json"
schedule = "cron(0 6 8 * ? *)"
skip-price = "0.68"
ssh-key = *******
```

### Carrier
H2O Wireless

### Email Notification Output

```
RattlesnakeOS Build FAILED
  Stack Name: ******
  Device: blueline
  Stack Version: 12.0.3
  Stack Region: us-east-2
  Instance Type: c5.4xlarge
  Instance Region: us-east-2
  Instance IP: 3.137.204.87
  Elapsed Time: 8hrs 4min 47sec
  Release: 20220104
  Tag: android-12.0.0_r25
  Build ID: SP1A.210812.016.A2
                         (/home/ubuntu/keys/blueline/platform)
Rewriting SYSTEM/build.prop:
[... 210 lines cut ...]
tmpfs            20M   60K   20M   1% /home/ubuntu/keys
/dev/loop4       62M   62M     0 100% /snap/core20/1270
/dev/loop5       52M   52M     0 100% /snap/snapcraft/6954
/dev/loop6       44M   44M     0 100% /snap/snapd/14295
/dev/loop7       56M   56M     0 100% /snap/core18/2253
/dev/loop8       68M   68M     0 100% /snap/lxd/21835
/dev/loop9       25M   25M     0 100% /snap/amazon-ssm-agent/4046
199G    /home/ubuntu/aosp
199G    total
 14:06:25 up  8:05,  0 users,  load average: 2.34, 6.31, 21.41
Completed 256.0 KiB/28.6 MiB (1.9 MiB/s) with 1 file(s) remaining
Completed 512.0 KiB/28.6 MiB (3.7 MiB/s) with 1 file(s) remaining
Completed 768.0 KiB/28.6 MiB (5.3 MiB/s) with 1 file(s) remaining
Completed 1.0 MiB/28.6 MiB (6.9 MiB/s) with 1 file(s) remaining 
Completed 1.2 MiB/28.6 MiB (8.4 MiB/s) with 1 file(s) remaining 
Completed 1.5 MiB/28.6 MiB (9.8 MiB/s) with 1 file(s) remaining 
Completed 1.8 MiB/28.6 MiB (11.3 MiB/s) with 1 file(s) remaining
Completed 2.0 MiB/28.6 MiB (12.7 MiB/s) with 1 file(s) remaining
Completed 2.2 MiB/28.6 MiB (14.0 MiB/s) with 1 file(s) remaining
Completed 2.5 MiB/28.6 MiB (15.4 MiB/s) with 1 file(s) remaining
Completed 2.8 MiB/28.6 MiB (16.7 MiB/s) with 1 file(s) r
```

### Full log
Let me know if you need it.

dan-v, 2022-01-09:
hi @IamJeffG - yeah that looks like it's probably a real issue; different device builds will use more or less space depending on what is required in the build and also the AOSP tree size is always growing, so it's possibly been on the verge of running out of space for a while and this month just put it over the edge. the link you provided is where the volume size is set. if you are able to edit that to something like 350GB, compile, deploy, and then do a build that would be helpful just to make sure it fixes the issue for you. if not, i can bump it and just put out a new release as an unverified fix.

IamJeffG, 2022-01-10:
Thanks -- I'll try this and report back.

commit e803c89f (2022-01-22): increase volume size from 300->350GB. closes #197
closed 2022-01-22 by dan-v